Qucs/qucs #687: TROUBLES WITH UNDEFINED MACROS

VitoSan92, 2017-05-16:
Hello everyone

I was working with a model in verilog A on Qucs 0.0.19 (on ubuntu 16.1), and this warning makes me crazy:

[image]

and these warnings make the compiler unsuccesfull:

make -f /usr/local/include/qucs-core/cpp2lib.makefile PREFIX="/usr/local" PROJDIR="/home/vito/.qucs/CNTFET_projects_prj" MODEL=cntfet_cap_GS_debug

g++  -std=gnu++11 -fPIC -shared -rdynamic -Wl,-soname,cntfet_cap_GS_debug.so -DHAVE_CONFIG_H -DQUCSADMS_DYLOAD -I./  -I"/usr/local"/include/qucs-core -o cntfet_cap_GS_debug.so cntfet_cap_GS_debug.cpp -L"/usr/local"/lib -Wl,-rpath -Wl,"/home/vito/.qucs/CNTFET_projects_prj" -lqucs -ldl
In file included from /usr/local/include/qucs-core/real.h:31:0,
                 from /usr/local/include/qucs-core/complex.h:29,
                 from /usr/local/include/qucs-core/component.h:37,
                 from cntfet_cap_GS_debug.CNTFET_cap_GS_debug.core.cpp:16,
                 from cntfet_cap_GS_debug.cpp:12:
./constants.h:11:1: error: stray â`â in program
 `ifdef CONSTANTS_VAMS
 ^
./constants.h:12:1: error: stray â`â in program
 `else
 ^
./constants.h:13:1: error: stray â`â in program
 `define CONSTANTS_VAMS 1
 ^
./constants.h:16:1: error: stray â`â in program
 `define M_E        2.7182818284590452354
 ^
./constants.h:17:1: error: stray â`â in program
 `define M_LOG2E    1.4426950408889634074
 ^
./constants.h:18:1: error: stray â`â in program
 `define M_LOG10E   0.43429448190325182765
 ^
./constants.h:19:1: error: stray â`â in program
 `define M_LN2      0.69314718055994530942
 ^
./constants.h:20:1: error: stray â`â in program
 `define M_LN10     2.30258509299404568402
 ^
./constants.h:21:1: error: stray â`â in program
 `define M_PI       3.14159265358979323846
 ^
./constants.h:22:1: error: stray â`â in program
 `define M_TWO_PI   6.28318530717958647652
 ^
./constants.h:23:1: error: stray â`â in program
 `define M_PI_2     1.57079632679489661923
 ^
./constants.h:24:1: error: stray â`â in program
 `define M_PI_4     0.78539816339744830962
 ^
./constants.h:25:1: error: stray â`â in program
 `define M_1_PI     0.31830988618379067154
 ^
./constants.h:26:1: error: stray â`â in program
 `define M_2_PI     0.63661977236758134308
 ^
./constants.h:27:1: error: stray â`â in program
 `define M_2_SQRTPI 1.12837916709551257390
 ^
./constants.h:28:1: error: stray â`â in program
 `define M_SQRT2    1.41421356237309504880
 ^
./constants.h:29:1: error: stray â`â in program
 `define M_SQRT1_2  0.70710678118654752440
 ^
./constants.h:34:1: error: stray â`â in program
 `define P_Q 1.6021918e-19
 ^
./constants.h:37:1: error: stray â`â in program
 `define P_C 2.997924562e8
 ^
./constants.h:40:1: error: stray â`â in program
 `define P_K 1.3806226e-23
 ^
./constants.h:43:1: error: stray â`â in program
 `define P_H 6.6260755e-34
 ^
./constants.h:46:1: error: stray â`â in program
 `define P_EPS0 8.85418792394420013968e-12
 ^
./constants.h:49:1: error: stray â`â in program
 `define P_U0 (4.0e-7 * `M_PI)
 ^
./constants.h:49:1: error: stray â`â in program
./constants.h:52:1: error: stray â`â in program
 `define P_CELSIUS0 273.15
 ^
./constants.h:54:1: error: stray â`â in program
 `endif
 ^
./constants.h:11:2: error: âifdefâ does not name a type
 `ifdef CONSTANTS_VAMS
  ^
In file included from cntfet_cap_GS_debug.cpp:12:0:
cntfet_cap_GS_debug.CNTFET_cap_GS_debug.core.cpp: In member function âvoid CNTFET_cap_GS_debug::calcVerilog()â:
cntfet_cap_GS_debug.CNTFET_cap_GS_debug.core.cpp:619:7: warning: division by zero [-Wdiv-by-zero]
 vgs=(1/0);
       ^
cntfet_cap_GS_debug.CNTFET_cap_GS_debug.core.cpp:26:11: error: expected unqualified-id before numeric constant
 #define n 1
           ^
cntfet_cap_GS_debug.CNTFET_cap_GS_debug.core.cpp:653:5: note: in expansion of macro ânâ
 int n = 0; (void) n;
     ^
cntfet_cap_GS_debug.CNTFET_cap_GS_debug.core.cpp:655:2: error: lvalue required as left operand of assignment
 n=indice1;
  ^
In file included from cntfet_cap_GS_debug.cpp:12:0:
cntfet_cap_GS_debug.CNTFET_cap_GS_debug.core.cpp:1393:7: warning: division by zero [-Wdiv-by-zero]
 vgs=(1/0);
       ^
/usr/local/include/qucs-core/cpp2lib.makefile:82: set di istruzioni per l'obiettivo "cntfet_cap_GS_debug.so" non riuscito
make: *** [cntfet_cap_GS_debug.so] Errore 1

How can I solve this? Help me please!!

guitorri, 2017-05-17:
You said by email that you fixed this error already:
```
cntfet_cap_GS_debug.CNTFET_cap_GS_debug.core.cpp:26:11: error: expected unqualified-id before numeric constant
 #define n 1
           ^
cntfet_cap_GS_debug.CNTFET_cap_GS_debug.core.cpp:653:5: note: in expansion of macro â��nâ��
 int n = 0; (void) n;
```

And the remaining error is about `ifdefs during the symbol generation, right?

VitoSan92, 2017-05-17:
Yes Mr. Torri. For a better explanation of the situation, this is the report of admsXml:

make -f /usr/include/qucs-core/va2cpp.makefile ADMSXML=/usr/bin/admsXml PREFIX=/usr MODEL=cntfet

# va2cpp - Creating C++ sources.

# BUG: must catch stray spaces in ADMSXML, adding extra quotation

"/usr/bin/admsXml" cntfet.va \

-I "/usr/include/qucs-core" \

-e "/usr/include/qucs-core/qucsVersion.xml" \

-e "/usr/include/qucs-core/qucsMODULEcore.xml" \

-e "/usr/include/qucs-core/qucsMODULEdefs.xml" \

-e "/usr/include/qucs-core/analogfunction.xml" \

-o cntfet

[info...] admsXml-2.3.4 (a508cf6) Feb 5 2016 13:34:51

[info...] [compact.vams:57]: macro `ifdef is undefined

[info...] [compact.vams:57]: macro `else is undefined

[info...] [compact.vams:57]: macro `endif is undefined

[warning] cntfet: device not handled by the adms qucs interface

[warning] please ensure extra code to be added to the interface

[info...] cntfet.core.cpp and cntfet.core.h: files created

[info...] cntfet.defs.h: file created

[info...] cntfet.cntfet.analogfunction.h created

[info...] cntfet.cntfet.analogfunction.cpp created

[info...] elapsed time: 0 (second)

[info...] admst iterations: 184733 (168300 freed)

and this is compiler one:

make -f /usr/include/qucs-core/cpp2lib.makefile PREFIX="/usr" PROJDIR="/home/vito/.qucs/CNTFET_project_2_prj" MODEL=cntfet

g++  -std=gnu++11 -fPIC -shared -rdynamic -Wl,-soname,cntfet.so -DHAVE_CONFIG_H -DQUCSADMS_DYLOAD -I./  -I"/usr"/include/qucs-core -o cntfet.so cntfet.cpp -L"/usr"/lib -Wl,-rpath -Wl,"/home/vito/.qucs/CNTFET_project_2_prj" -lqucs -ldl

The error comes only with symbol generation.

A: Qucs/qucs
Cc: SANTORSOLA VITO; Author

VitoSan92, 2017-05-18:
Is very important for me to fix this trouble. So I need for an answer.

[5x1000]

guitorri, 2017-05-18:
The issue is related to:

```
"/usr/bin/admsXml" cntfet.va \
-I "/usr/include/qucs-core" \
-e "/usr/include/qucs-core/qucsVersion.xml" \
-e "/usr/include/qucs-core/qucsMODULEcore.xml" \
-e "/usr/include/qucs-core/qucsMODULEdefs.xml" \
-e "/usr/include/qucs-core/analogfunction.xml" \
-o cntfet
[info...] admsXml-2.3.4 (a508cf6) Feb 5 2016 13:34:51
[info...] [compact.vams:57]: macro `ifdef is undefined
[info...] [compact.vams:57]: macro `else is undefined
[info...] [compact.vams:57]: macro `endif is undefined
```

And again here:
https://cloud.githubusercontent.com/assets/28738493/26163142/038dafd6-3b29-11e7-9d42-9596bf560e5a.png

You seem to have a function with two inputs on your `compact.vams`.
It might be needed to add a dummy `mapply_binary` to the `*symbol.xml` file.

It normally can handle `` `ifdef ``, I don't know why it is not handling in your case.
Is there something being defined? Such as `` `ifdef _something_``?

Other than that, I guess I would need to see the source code to check what is going on. Is that something you can share?

I know your work is very important... we all need answers :)

felix-salfelder, 2017-05-18:
undefined `` `ifdef`` does not really make any sense to me. can you recheck the file encoding? could you post the offending lines plus some context?

does admsXml complain when parsing `constants.vams` standalone?

VitoSan92, 2017-05-18:
Hi

I tried again to make compilation, and compiler troubles come back! (see image "situation").

Those are admsXml report:

make -f /usr/local/include/qucs-core/va2cpp.makefile ADMSXML=/usr/bin/admsXml PREFIX=/usr/local MODEL=cntfet

# va2cpp - Creating C++ sources.

"/usr/bin/admsXml" cntfet.va \

-I "/usr/local/include/qucs-core" \

-e "/usr/local/include/qucs-core/qucsVersion.xml" \

-e "/usr/local/include/qucs-core/qucsMODULEcore.xml" \

-e "/usr/local/include/qucs-core/qucsMODULEdefs.xml" \

-e "/usr/local/include/qucs-core/analogfunction.xml" \

-o cntfet

[info...] admsXml-2.3.4 (a508cf6) Feb 5 2016 13:34:59

[info...] [compact.vams:55]: macro `ifdef is undefined

[info...] [compact.vams:55]: macro `else is undefined

[info...] [compact.vams:55]: macro `endif is undefined

[warning] cntfet: device not handled by the adms qucs interface

[warning] please ensure extra code to be added to the interface

[info...] cntfet.core.cpp and cntfet.core.h: files created

[info...] cntfet.defs.h: file created

[info...] cntfet.cntfet.analogfunction.h created

[info...] cntfet.cntfet.analogfunction.cpp created

[info...] elapsed time: 0 (second)

[info...] admst iterations: 184748 (168300 freed)

and compiler one:

make -f /usr/local/include/qucs-core/cpp2lib.makefile PREFIX="/usr/local" PROJDIR="/home/vito/.qucs/CNTFET_projects_prj" MODEL=cntfet

g++ -std=gnu++11 -fPIC -shared -rdynamic -Wl,-soname,cntfet.so -DHAVE_CONFIG_H -DQUCSADMS_DYLOAD -I./ -I"/usr/local"/include/qucs-core -o cntfet.so cntfet.cpp -L"/usr/local"/lib -Wl,-rpath -Wl,"/home/vito/.qucs/CNTFET_projects_prj" -lqucs -ldl

In file included from /usr/local/include/qucs-core/real.h:31:0,

from /usr/local/include/qucs-core/complex.h:29,

from /usr/local/include/qucs-core/component.h:37,

from cntfet.cntfet.core.cpp:16,

from cntfet.cpp:12:

./constants.h:11:1: error: stray â€˜`â€™ in program

`ifdef CONSTANTS_VAMS

^

./constants.h:12:1: error: stray â€˜`â€™ in program

`else

^

./constants.h:13:1: error: stray â€˜`â€™ in program

`define CONSTANTS_VAMS 1

^

./constants.h:16:1: error: stray â€˜`â€™ in program

`define M_E 2.7182818284590452354

^

./constants.h:17:1: error: stray â€˜`â€™ in program

`define M_LOG2E 1.4426950408889634074

^

./constants.h:18:1: error: stray â€˜`â€™ in program

`define M_LOG10E 0.43429448190325182765

^

./constants.h:19:1: error: stray â€˜`â€™ in program

`define M_LN2 0.69314718055994530942

^

./constants.h:20:1: error: stray â€˜`â€™ in program

`define M_LN10 2.30258509299404568402

^

./constants.h:21:1: error: stray â€˜`â€™ in program

`define M_PI 3.14159265358979323846

^

./constants.h:22:1: error: stray â€˜`â€™ in program

`define M_TWO_PI 6.28318530717958647652

^

./constants.h:23:1: error: stray â€˜`â€™ in program

`define M_PI_2 1.57079632679489661923

^

./constants.h:24:1: error: stray â€˜`â€™ in program

`define M_PI_4 0.78539816339744830962

^

./constants.h:25:1: error: stray â€˜`â€™ in program

`define M_1_PI 0.31830988618379067154

^

./constants.h:26:1: error: stray â€˜`â€™ in program

`define M_2_PI 0.63661977236758134308

^

./constants.h:27:1: error: stray â€˜`â€™ in program

`define M_2_SQRTPI 1.12837916709551257390

^

./constants.h:28:1: error: stray â€˜`â€™ in program

`define M_SQRT2 1.41421356237309504880

^

./constants.h:29:1: error: stray â€˜`â€™ in program

`define M_SQRT1_2 0.70710678118654752440

^

./constants.h:34:1: error: stray â€˜`â€™ in program

`define P_Q 1.6021918e-19

^

./constants.h:37:1: error: stray â€˜`â€™ in program

`define P_C 2.997924562e8

^

./constants.h:40:1: error: stray â€˜`â€™ in program

`define P_K 1.3806226e-23

^

./constants.h:43:1: error: stray â€˜`â€™ in program

`define P_H 6.6260755e-34

^

./constants.h:46:1: error: stray â€˜`â€™ in program

`define P_EPS0 8.85418792394420013968e-12

^

./constants.h:49:1: error: stray â€˜`â€™ in program

`define P_U0 (4.0e-7 * `M_PI)

^

./constants.h:49:1: error: stray â€˜`â€™ in program

./constants.h:52:1: error: stray â€˜`â€™ in program

`define P_CELSIUS0 273.15

^

./constants.h:54:1: error: stray â€˜`â€™ in program

`endif

^

./constants.h:11:2: error: â€˜ifdefâ€™ does not name a type

`ifdef CONSTANTS_VAMS

^

/usr/local/include/qucs-core/cpp2lib.makefile:82: set di istruzioni per l'obiettivo "cntfet.so" non riuscito

make: *** [cntfet.so] Errore 1

I send the codes of interest in this email.

However Thank you for answers!

https://cloud.githubusercontent.com/assets/28738493/26163142/038dafd6-3b29-11e7-9d42-9596bf560e5a.png

[https://cloud.githubusercontent.com/assets/28738493/26163142/038dafd6-3b29-11e7-9d42-9596bf560e5a.png]

[5x1000]

VitoSan92, 2017-05-18:
How can I add a dummy mapply_binary to the *symbol.xml file?? I don't have a great knowledge upon this question.

guitorri, 2017-05-22:
AdmsXml  parses the VerilogA code and generates C++ code.
For doing that a set of templates (XML files) are given to tell admsXml how to translate the parsed file into another format.

For symbol generation, the module name, ports and list off parameters are needed.

During the code generation, `qucsMODULEguiJSONsymbol.xml` is encountering a binary operator, something with to input arguments, such as `atan2(x,y)` . 

It does not know how to handle it. Hence the fail.

If you look into `qucsMODULEguiJSONsymbol.xm` there a `:mapply_unary`, but not a `:mapply_binary`, see:

```
<admst:template match=":mapply_unary">
```

You can find an example of `:mapply_binary` in `qucsMODULESdef.xml`. You can try to copy section to see if it works.

Still I don't understand where the binary operator is located in your model. Is there any function with two inputs on your parameter list?

guitorri, 2017-10-04:
No test case provided. Closing for now.

closed 2017-10-04 by guitorri