Qucs/qucs #689: schematic and ordering

felix-salfelder, 2017-06-30:
there are many ways to order schematic items. most will be not visible/relevant on the user end.

one problem seems to be, that some simulator backend or netlister will need items in a particular order. perhaps there are more.

* should we try to generalize and provide all thinkable ways in the library?
* maybe even change the way things are stored (group them)
* or should we just provide iterator access to the schematics?

all 3 have upsides and downsides. lets discuss this, before we implement it.

ra3xdh, 2017-06-30:
> in the library

What libraries do you mean? Is it traditional Qucs libraries that are accessed through LibComp?

felix-salfelder, 2017-06-30:
> What libraries

any of the qucs libraries. obviously, the schematic code is slightly messy, and it's hard to see any (library) structure there. look at how gnucap is organised, it's much easier to see a pattern there. without that pattern...  think of it the library as the "non-plugin code", which is the part of the code that does not need to be turned into plugins.

above, i was referring particularly to the part that implements the containers for the stuff you see in your schematic, and the interface for accessing that.

closed 2021-02-21 by felix-salfelder