Qucs/qucs #697: Build Error: qrc_qucs.cpp: No such file or directory

aviadb, 2017-07-14:
Building qucs from latest "develop" branch
fails with error
g++: error: qrc_qucs.cpp: No such file or directory
g++: fatal error: no input files
compilation terminated.
make[4]: *** [Makefile:641: qrc_qucs.o] Error 1

in3otd, 2017-07-14:
cannot reproduce here - also the automatic builds on Travis and Appveyor were fine.
Which OS and build system (CMake / autotools) are you using?

If you do `rcc -v` do you have some meaningful output? Do you happen to have an `rcc-qt4` or similar on your system also?

If you used `./configure` could you check what was printed exactly on the line saying 
```
checking for rcc...
```

aviadb, 2017-07-16:
Manjaro linux
Cmake 3.8.2
rcc is not available on my system as executable.
checking for rcc ... no.

Searching for rcc with my package manager gives the following results:
librcc 0.2.12 - Charset conversion library
pyside-tools-common 0.2.15 - PySide lupdate, rcc, and uic development tools
pyside-tools-common 0.2.15 - for python 3
pyside-tools-common 0.2.15 - for python 2

nothing related to qt

in3otd, 2017-07-16:
so the issue comes from the fact that the `rcc` program, which comes with Qt, is not found, quite strange, it should be available somewhere.
I forgot to ask also for the output of the `configure` lines
```
checking for moc-qt4...
checking for moc...
checking for uic-qt4...
checking for uic...
```
did you check if `rcc-qt4` works? (e.g. `rcc-qt4 -v`) Googling around it seems that Manjaro uses that name for the Qt4 `rcc`, which would explain the issue: for some reason we do not check if `rcc` is available with that name (but do so for other Qt utilities)

aviadb, 2017-07-17:
checking for moc-qt4... /usr/bin/moc-qt4
no line for checking for moc...
checking for uic-qt4... /usr/bin/uic-qt4
no line for checking for uic...

rcc-qt4 -v
prints: Qt Resource Compiler version 4.8.7

:S

in3otd, 2017-07-17:
thanks, this confirms that the issue comes from our `configure` script, which [looks only for `rcc`](https://github.com/Qucs/qucs/blob/master/qucs/configure.ac#L710) and not also for `rcc-qt4`, as done for the other Qt utilities. I'll push a fix later.

felix-salfelder, 2017-07-17:
maybe plain

     - AC_PATH_PROGS(RCC, rcc, :, [$QT_DIR/bin:$PATH])
     + AC_PATH_PROGS(RCC, rcc-qt4 rcc, :, [$QT_DIR/bin:$PATH])

will do it.

if the rest of the build system works as intended, an invocation like `make RCC=rcc-qt4` should easily work around that.

in3otd, 2017-07-17:
yes, that's what I was going to do :grin: , plus the `AC_MSG_ERROR()` blob following it, which also should have been there in the first place to tell the user where the problem was.

aviadb, 2017-07-17:
looks like the rcc-qt package is installed on my system, however the build still fails.
Is it because I do not have the rcc package?

in3otd, 2017-07-17:
you mean it still fails after modifying the [configure.ac line](https://github.com/Qucs/qucs/blob/master/qucs/configure.ac#L710) as suggested by felix ?
Or, if the question was about why the build fails while having the needed application installed is that that application can be installed with different names, usually `rcc` or `rcc-qt4` and we check just for the first one, currently - likely for historical reasons which are no longer valid now.

aviadb, 2017-07-17:
Actually the ./configure passes
the failure is happening after executing the make cmd.

aviadb, 2017-07-17:
probably qrc_qucs.cpp is not included because thats what cause the compiler to fail.

g++: error: qrc_qucs.cpp: No such file or directory
g++: fatal error: no input files
compilation terminated.
make[4]: *** [Makefile:641: qrc_qucs.o] Error 1

in3otd, 2017-08-06:
this should be now fixed in the `develop` branch by #699

closed 2017-08-06 by in3otd