Qucs/qucs #698: transient simulation sometimes takes too long

bergmannm, 2017-07-17:
Hi,
I'm trying to simulate negative rail generation of http://www.electronics-lab.com/project/0-30-vdc-stabilized-power-supply-with-current-control-0-002-3-a/ .  I've uploaded simulated circuit at 
http://imgur.com/a/NX7pX    and https://pastebin.com/wHpnzW1X . The problem is that simulation time depends quite a bit  on selected values. I've tried to change value of R6, all other settings remained the same.

With R6=500 Ohm -> less than a second
With R6=300 Ohm  -> about 5 seconds
With R6=400 Ohm  -> simulation was not finished even after 5 minutes.

Is there a way to speed up simulations ?

ra3xdh, 2017-07-17:
Are you using Qucsator? Yes, it is well known bug. Qucsator may hang when some component goes into switch mode. See also #41. Qucsator is more suitable for RF circuits simulation in frequency domain. There was some attempts to fix its performance in time domain. 

Use SPICE simulation kernels to simulate such circuits. My Qucs-S + Ngspice can give a good result. See https://ra3xdh.github.io/ for more details. I hope, we will see Qucs-S and Qucs merged in the next year. You may also like experimental Ngspice support in KiCAD.

ra3xdh, 2017-07-17:
Closed as duplicate of #41.

closed 2017-07-17 by ra3xdh

felix-salfelder, 2017-07-17:
for completeness, let me add that there is gnucsator and that it provides a qucsator drop in replacement.
gnucsator is gnucap based and implements a real ("post-spice") transient solver.

unfortunately, some of the components are still missing, and only transient simulation is wrapped.
native gnucap support will be available as soon as we have modular simulator drivers for qucs.

ombra32, 2021-05-29:
I'm having the same issue 4 years later with Qucs 0.0.20 on Ubuntu 20.04.
In a simple AC generator and R divider circuit, the simulation is stuck at 0%.
I can't use Qucs-s because in another circuit I need a triac and Qucs-s lacks of some non linear components (see the related [issue](https://github.com/ra3xdh/qucs_s/issues/63)).

[image]

ra3xdh, 2021-05-29:
What is the frequency of the AC source? Looks like you have accidentally left the default value 1GHz unchanged. In such case the simulation with qucsator will stuck.

ombra32, 2021-05-29:
> What is the frequency of the AC source? Looks like you have accidentally left the default value 1GHz unchanged. In such case the simulation with qucsator will stuck.

You are right! Congratulations!
I'm very sorry, I'm new to Qucs.
Maybe having the AC source frequency set to 50Hz by default, like some other circuit simulators do, can help beginners and save developers from boring answers..

Regards,
Marco

Rmano, 2021-05-31:
Well, 100~ms at 1 GHz is *quite a lot* of periods to be simulated... maybe it's not stuck ;-)

ombra32, 2021-06-02:
I'm stuck in another circuit now: a full diode bridge with an AC source and a time controlled switch.
As you stated, Qucsator seems not so good for switching simulations, but time controlled switch is not present in Qucs-s.

[image]

ra3xdh, 2021-06-02:
You may use the relay instance controlled by rectangular pulse source. This device is available for Qucs-S and SPICE-compatible.

Rmano, 2021-06-02:
@ombra32  it would be useful if you could share the `.SCH` file (you can zip it and drop it into an issue here) so that we could check...

ombra32, 2021-06-02:
> @ombra32 it would be useful if you could share the `.SCH` file (you can zip it and drop it into an issue here) so that we could check...

No problem, it's a very basic circuit
[dimmerID.zip](https://github.com/Qucs/qucs/files/6585015/dimmerID.zip)

Rmano, 2021-06-02:
Well, I received errors about Ron and Roff with your circuits --- those are critical parameters for the convergence of the simulation...
After tweaking them a bit, I have this (run time around 3 seconds):

[image]

Rmano, 2021-06-02:
> Many thanks Romano.
> How can I receive this errors too? Is there an advanced log or simulation mode to enable?

No, they were just in the simulation windows. Which version are you using? Mine is this one 0.0.20-rc2 (the appexec thingy I made)

ombra32, 2021-06-02:
Qucs 0.0.20.201022-2 from launchpad repo on Ubuntu 20.04
simulation stucks as my previous image without messages

[image]

Rmano, 2021-06-02:
...no idea, I didn't even know there was a working repo for 20.04. The project is in bad need of C++ developers to port it from Qt4 to Qt5-6, and the version I packed is from June 2020...

See here: https://github.com/Qucs/qucs/issues/1006#issuecomment-643792968 and here:  https://rmano.github.io/qucsAppImagesBuild/  --- but no guarantees at all, it is just a stop-gap for now.

ombra32, 2021-06-02:
> ...no idea, I didn't even know there was a working repo for 20.04. The project is in bad need of C++ developers to port it from Qt4 to Qt5-6 (I would like to be able to help, and the existing developers are doing all that they can!), and the version I packed is from June 2020...

I've installed from here:
[http://qucs.sourceforge.net/install.html#install_ubuntu](http://qucs.sourceforge.net/install.html#install_ubuntu)
I think is the same for all Ubuntu versions

> See here: [#1006 (comment)](https://github.com/Qucs/qucs/issues/1006#issuecomment-643792968) and here: https://rmano.github.io/qucsAppImagesBuild/ --- but no guarantees at all, it is just a stop-gap for now.

Ok, I'll give a try..many thanks!

Rmano, 2021-06-02:
> Ok, I'll give a try..many thanks!

Well, the PPA seems more recent. I really don't know, you can just try. But does it work with my modifications to Ron and R0ff? If so, I do not think it's a big problem. Ron and Roff of switches are a PITA for every simulator...

ombra32, 2021-06-03:
With qucs appimage 0.0.20-rc2 I get the message you mentioned, but with 0.0.20 version from PPA still no message.

[image]

> But does it work with my modifications to Ron and R0ff? If so, I do not think it's a big problem. Ron and Roff of switches are a PITA for every simulator...

 Yes, changing Roff only, it works with both versions.