Qucs/qucs #700: Subcircuit is not placed in the main circuit

ZapodovnikovKI, 2017-07-19:
A subcircuit is not placed in the main circuit. Tested in different OS (WinXP, Win7 32 and 64). 
What's the problem?

in3otd, 2017-07-19:
we will need some more details to understand what's wrong; I assume you are using the latest code from the `develop` branch.
How are you trying to place the subcircuit, by drag and drop from the *Content* tab in the *Main Dock* on the left side? Can you see the subcicruit symbol being dragged? Does your subcircuit have at least one "Subcircuit Port* element in its schematic? (otherwise is not considered to be a subcircuit)

ZapodovnikovKI, 2017-07-19:
I'm trying to drag the subcircuit from the tab "content" in the Main left side (so I'm doing 16 and 18, QUCS). Subschema not drags. My subcircuits have at least one "Subcircuit's port. I actively used subcircuits in previous versions. The old rules do not work.

in3otd, 2017-07-19:
ok, very strange.
Do you see the number of ports of the subcircuit in the *Note* field, as shown here http://qucs-help.readthedocs.io/en/latest/subcircuit.html
If you double click on the subcircuit name in the *Content* tab can you open its schematic?

Can you share a project showing the issue to see if it works here?

ZapodovnikovKI, 2017-07-20:
I see the Number of ports of the subcircuit in the note field. 
I open the subcircuit schematic by double-clicking on its name. This happens in version qucs-0.0.19-RC6 (https://github.com/ra3xdh/qucs/releases/tag/0.0.19S-rc6). 
Today found qucs-0.0.19-Win32-mingw482-ASKO-freehdl (https://sourceforge.net/projects/qucs/files/latest/download?source=typ_redirect ). 
Here subcircuit "dragged" not usually. One click on subcircuit that takes pointer of mouse on field. The second click and subcircuit is fixed on field. 
In previous versions took schematic or other circuit elements by clicking left mouse button and literally dragging on the field without releasing button. 
The severity of my problems have disappeared. I used the unofficial version 19s-RC6. 
Left the actual change  operation "drag and drop" in the official version 19-Win32-mingw482-ASKO-freehdl.

ZapodovnikovKI, 2017-07-20:
It's my test ...
[zProb19s_prj.zip](https://github.com/Qucs/qucs/files/1161240/zProb19s_prj.zip)

in3otd, 2017-07-20:
it could be that in the old Qucs versions it was possible to place a subcircuit by a real drag and drop, while now you need to first click on the subcircuit name, release the mouse button and only then you see the subcircuit simbol "dragged" to the schematic; the *Content* tab all the tabs on the left side were completely rewritten some time ago and it seems we lost the drag and drop functionality there. 

I quickly tried to re-enable it but there is some more work to do there...

ZapodovnikovKI, 2017-07-22:
There are still a few GUI bugs. 
Take out the MUL (or VSUM, VSub, etc.) of the Ideal Library components (Tools) drag and drop(!). Connect the conductors. Turn the item. Insights don't connect. There is a displacement of the grid. 
Still an old error in Sqrt. We put on field and in image find third conclusion. The item is not workable. Error.

andresmmera, 2017-07-22:
I reckon that you may solve the grid stuff by doing right-click on the component and selecting "Set on Grid (CTRL+U)"

ZapodovnikovKI, 2017-07-23:
Ctrl+U works. It's fast  movement to the beauty of image schema. Thank you!

closed 2018-07-19 by guitorri