Qucs/qucs #702: Incorrect formula for conductive losses in CPW

edumur, 2017-07-26:
In the cpwline.cpp it looks like the conductive losses are calculated as if the result of the formula from Ghione is in Neper/meter.
However in the Gupta, author claims that the formula is in Db/meter
[image]

How is right?

in3otd, 2017-07-26:
> Who is right?

both. (actually the formula you reported above has an error, but not related to the unit of loss)

edumur, 2017-07-26:
That's interesting.
Do you mean that:
1. The formula used in the code is correct to give a result in Neper/meter.
2. The formula in Gupta is incorrectly copied from the source [36]:
[image]

Thanks for very fast reply ;) .

in3otd, 2017-07-26:
1. yes
2. yes

but checking the sources/references I saw that the Qucs documentation is confusing, since it uses different units for the same quantity in the same chapter, see http://qucs.sourceforge.net/tech/node86.html

edumur, 2017-07-26:
Well this is where I started my journey about the formula.
If the reference [39], Gupta in the documentation, is wrong, don't you think that should be made clear in the documentation?

in3otd, 2017-07-26:
reference [39] is *K. C. Gupta, R. Garg, I. J. Bahl, and P. Bhartia, Microstrip Lines and Slotlines, 2nd ed.Artech House, Inc., 1996.* which, curiously enough, has the correct expression. I guess the one you quoted above is from the third edition, or?

edumur, 2017-07-26:
Good catch!
My bad, it effectively comes from the third edition.

I would then say that everything seems correct to me.
If you agree, I think this thread can be closed.

And thanks again!

in3otd, 2017-07-27:
thanks - I reopened the issue, changing the title to remind to clarify the loss units in the CPW docs