Qucs/qucs #710: Qucsator crashes on trivial netlists

pienjo, 2017-09-09:
Hello,

Qucsator (from unmodified qucs-0.0.19 sources) suddenly started crashing on my system (void linux, Linux 4.12.10, x86_64, glibc 2.26), with a segmentation fault in qucs::net::containsAnalysis(), line 224. It doesn't really matter what circuit I feed it; my crash dump below is using the supplied netlist, consisting of a voltage source and a resistor.

[netlist.txt](https://github.com/Qucs/qucs/files/1289880/netlist.txt)

Output:

[martijnb@tinkerbell .libs]$ ./qucsator -i ~/.qucs/netlist.txt 
project location: 
modules to load: 0
factorycreate.size() is 0
factorycreate has registered:
parsing netlist...
checking netlist...
netlist content
      1 DC instances
      1 R instances
      1 Vdc instances
creating netlist...
Segmentation fault
[martijnb@tinkerbell .libs]$ 

gdb stack trace:

#0  0x00007ffff78b855a in std::list<qucs::analysis*, std::allocator<qucs::analysis*> >::begin (this=0x21)
    at /usr/include/c++/6.3/bits/stl_list.h:841
#1  0x00007ffff78b6fb4 in qucs::net::containsAnalysis (this=0x61ff10, child=0x6214e0, type=1) at net.cpp:224
#2  0x00007ffff78b7870 in qucs::net::sortChildAnalyses (this=0x61ff10, parent=0x6226e0) at net.cpp:355
#3  0x00007ffff78b7768 in qucs::net::orderAnalysis (this=0x61ff10) at net.cpp:344
#4  0x00007ffff78b7144 in qucs::net::runAnalysis (this=0x61ff10, err=@0x7fffffffe61c: 0) at net.cpp:249
#5  0x00000000004049a5 in main (argc=3, argv=0x7fffffffe808) at ucs.cpp:251

Relevant source:

220     /* Looks recursively for a type of analysis. */                                                  
221     int net::containsAnalysis (analysis * child, int type) {                                      
222       ptrlist<analysis> * alist = child->getAnalysis ();                                            
223       if(alist != nullptr) {                                                                                          
224         for (auto *a : *alist) {                                                                                    
225           if (a->getType () == type)                                                                          
226             return 1;                                                                                                  
227           else if (a->getType () == ANALYSIS_SWEEP)                                                                         

228             return containsAnalysis (a, type);                                                           
229         }                                                                                                                  
230       }                                                                                                                    
231       return 0;                                                                                                            
232     }  

"alist" is obviously hosed:

(gdb) print alist 
$1 = (qucs::ptrlist *) 0x21
(gdb) 

but that's as far as my gdb skills get me - Debugging c++ with gdb or ddd is tedious enough when I know what I should be looking for, but I don't :( However, I have a core dump available, if you want, and I'm willing to set breakpoints and inspect data, if you tell me where :)

What's peculiar is that qucsator used to work just fine until recently - it hasn't been updated itself, but glibc *has* been recently updated. I'll try to double check that. And no, rebuilding qucs from sources doesn't help.

Any help would be appreciated.

Kind regards,
Martijn.

pienjo, 2017-09-09:
OK, that's confirmed: Forcibly downgrading to glibc-2.25 "fixes" the problem, but leaves the rest of my system in a broken state :-/

pienjo, 2017-09-09:
I think I found the problem, it's actually a level higher, in net::sortChildAnalyses():
// This function sorts the analyses of the given parent analysis.
void net::sortChildAnalyses (analysis * parent) {
  ptrlist<analysis> * alist = parent->getAnalysis ();
  if (alist != nullptr) {
    for (auto *a: *alist) {
      if (a->getType () == ANALYSIS_DC
	  || containsAnalysis (a, ANALYSIS_DC)) {
	parent->delAnalysis (a);
	parent->addAnalysis (a);
      }
    }
  }
}

The problem are the calls to addAnalysis() and especially delAnalysis(): They modify the very same list alist points to. The call to delAnalysis() is particularly dangerous here, as the 'remove' contained within invalidates the hidden iterator of the range based for, as it is removing the current iteration item. This triggers undefined behaviour - and apparently, "undefined behaviour" means crashing during the next iteration while using glibc-2.26, while glibc-2.25 is doing something less visible..

I'll try and come up with a patch.

closed 2017-10-04 by guitorri

mathieu-lacage, 2021-10-21:
It is kind of sad that this specific fix appears to not have made it yet to a stable release so, all qucs binaries installed from linux distribution packages are based on the 0.0.19 version which will simply crash out of the box on the hello world example in the documentation (resistor divider). 

I see a couple of bug reports in distribution bug trackers that appear to relate to the exact same bug:

https://bugzilla.redhat.com/show_bug.cgi?id=1891255
https://bugzilla.redhat.com/show_bug.cgi?id=2008299

Rebuilding from source the 0.0.20rc2 version works around this problem.