Qucs/qucs #711: qucs-0.0.19 fails to build - configure failed while checking for Qt library

1greyghost, 2017-09-09:
(This is on a new install of Linux Mint 18.2)

I did the following: 

ADMS
- downloaded and unpacked adms-2.3.6
- sudo apt-get install build-essential
- sudo apt-get install automake libtool gperf flex bison
- sudo apt-get install libxml2 libxml2-dev
- sudo apt-get install libgd-perl
- sudo sh bootstrap.sh
- sudo ./configure --enable-maintainer-mode
- sudo make install

Great adms made!!

QUCS
- downloaded and unpacked qucs
- sudo apt-get install libqt4-dev libqt4-qt3support
- sudo apt-get install texlive-font-utils octave-epstk
- installed qt4-dev-tools via synaptic 
- sudo ./configure

Bummer, the configure failed.  below are the last few lines of the configure process:

    checking for X... no
    checking for Qt headers... found in /usr/include/qt4 -I/usr/include/qt4/Qt -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtSvg -I/usr/include/qt4/QtXml -I/usr/include/qt4/QtScript -I/usr/include/qt4/Qt3Support
    checking for Qt... 4 (multi-threaded)
    checking for Qt library... configure: error: not found
    configure: error: ./configure failed for qucs

Any thoughts?

felix-salfelder, 2017-09-09:
checking for X... no

looks strange to me. no X on Mint?

    checking for Qt library... configure: error: not found

so you did instal some Qt libraries? which/how?
configure should have written more details to config.log,  can you show us the relevant chunk?

1greyghost, 2017-09-11:
felix, below is the config.log:

This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by qucs configure 0.0.19, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ./configure 

## --------- ##
## Platform. ##
## --------- ##

hostname = Elfworks
uname -m = x86_64
uname -r = 4.10.0-33-generic
uname -s = Linux
uname -v = #37~16.04.1-Ubuntu SMP Fri Aug 11 14:07:24 UTC 2017

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin
PATH: /snap/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:1904: checking build system type
configure:1918: result: x86_64-unknown-linux-gnu
configure:1938: checking host system type
configure:1951: result: x86_64-unknown-linux-gnu
configure:1988: checking for a BSD-compatible install
configure:2056: result: /usr/bin/install -c
configure:2067: checking whether build environment is sane
configure:2122: result: yes
configure:2273: checking for a thread-safe mkdir -p
configure:2312: result: /bin/mkdir -p
configure:2319: checking for gawk
configure:2335: found /usr/bin/gawk
configure:2346: result: gawk
configure:2357: checking whether make sets $(MAKE)
configure:2379: result: yes
configure:2408: checking whether make supports nested variables
configure:2425: result: yes
configure:2611: checking for g++
configure:2627: found /usr/bin/g++
configure:2638: result: g++
configure:2665: checking for C++ compiler version
configure:2674: g++ --version >&5
g++ (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

[... 261 lines cut ...]
host_os='linux-gnu'
host_vendor='unknown'
htmldir='${docdir}'
includedir='${prefix}/include'
infodir='${datarootdir}/info'
install_sh='${SHELL} /home/bobby/Downloads/qucs-0.0.19/install-sh'
libdir='${exec_prefix}/lib'
libexecdir='${exec_prefix}/libexec'
localedir='${datarootdir}/locale'
localstatedir='${prefix}/var'
mandir='${datarootdir}/man'
mkdir_p='$(MKDIR_P)'
oldincludedir='/usr/include'
pdfdir='${docdir}'
prefix='/usr/local'
program_transform_name='s,x,x,'
psdir='${docdir}'
runstatedir='${localstatedir}/run'
sbindir='${exec_prefix}/sbin'
sharedstatedir='${prefix}/com'
subdirs=' qucs qucs-core qucs-doc'
sysconfdir='${prefix}/etc'
target_alias=''

configure: exit 1

felix-salfelder, 2017-09-11:
> subdirs=' qucs qucs-core qucs-doc'

actually, (for some reason) there are two nested `qucs` packages. one consists of 3 packages related to qucs, the other is the qucs gui (the one that failed to configure). you find the logfile for the latter in `$builddir/qucs/config.log`. i guess your builddir is `/home/bobby/Downloads/qucs-0.0.19/`

(sorry for the confusion)

plomlompom, 2017-09-18:
I've had a similar issue (same "checking for Qt library... configure: error: not found") on the current Debian stable. In my case, the solution was to also install the libqt4-opengl-dev package, despite it not being listed in the README's list of dependencies.

felix-salfelder, 2017-09-18:
@plomlompom thanks.

did libqt4-opengl-dev pull in libqt4-opengl? maybe that one is somehow checked for (or required, but why?).

1greyghost, 2017-09-18:
plomlompom (Christian), YOU ARE THE MAN!

Installing 'libqt4-opengl-dev', allowed me to configure, with an exit value of 0.  Now I am in the process of making and installing.
--

Whoops, I noticed a lot of warnings about deprecated lines of code during the 'make and install', and I ended up getting an error.

The last several lines of the 'make and install' are, as follows:

    GEAR1 error constants: 2-step, order 1, error constant 1
            0   0   1
    GEAR2 error constants: 3-step, order 2, error constant 1
            0   0   0   1
    GEAR3 error constants: 4-step, order 3, error constant 1
            0.00000   0.00000   0.00000  -0.00000   1.00000
    GEAR4 error constants: 5-step, order 4, error constant 1
            0.00000   0.00000   0.00000  -0.00000   0.00000   1.00000
    GEAR5 error constants: 6-step, order 5, error constant 1
            0.00000   0.00000   0.00000   0.00000   0.00000  -0.00000   1.00000
    GEAR6 error constants: 7-step, order 6, error constant 1
            0.00000   0.00000   0.00000   0.00000   0.00000  -0.00000   0.00000   1.00000
    /usr/bin/octave --path . coplanarzl.m
    Graphic reduced !
    export TEXINPUTS=.:./..:./..:; : -interaction=nonstopmode technical.tex
    export BIBINPUTS=.:; /usr/bin/bibtex technical
    I couldn't open file name `technical.aux'
    Makefile:585: recipe for target 'technical.dvi' failed
    make[2]: *** [technical.dvi] Error 1
    make[2]: Leaving directory '/home/bobby/Downloads/qucs-0.0.19/qucs-doc/technical'
    Makefile:383: recipe for target 'install-recursive' failed
    make[1]: *** [install-recursive] Error 1
    make[1]: Leaving directory '/home/bobby/Downloads/qucs-0.0.19/qucs-doc'
    Makefile:381: recipe for target 'install-recursive' failed
    make: *** [install-recursive] Error 1

plomlompom, 2017-09-20:
@felix-salfelder: I just tried with mere libqt4-opengl, but that only failed with `checking for Qt library... configure: error: not found` again.

@1greyghost: Glad it helped a bit at least. The error you encounter also looks familiar. I think I overcame it by running `./configure` with the `--disable-doc` flag.

1greyghost, 2017-09-22:
@plomlompom: running './configure --disable-doc' did the trick.  I had no trouble with the 'make' and everything installed correctly.  Thanks!

closed 2017-10-08 by guitorri