Qucs/qucs #714: (qucsator) of user 1000 dumped core

sitilge, 2017-09-21:
I followed the Getting Started guide. Reading the `journalctl` it outputs
````
(qucsator) of user 1000 dumped core
````
and the Qucs Simulation Messages window outputs
````
ERROR: Simulator crashed!
````
Running `Qucs 0.0.19` on `arch 4.12.13`

guitorri, 2017-10-04:
Can you provide the .sch file so we can try to reproduce?

extrasharp, 2017-11-25:
Hello! I'm getting the same error under Arch
This is what journalctl says

```
Nov 25 00:15:01 kiwi-strawberry systemd-coredump[15984]: Process 15982 (qucsator) of user 1000 dumped core.
                                                         
                                                         Stack trace of thread 15982:
                                                         #0  0x00007fa45f933348 _ZN4qucs3net17sortChildAnalysesEPNS_8analysisE (libqucs.so.0)
                                                         #1  0x00007fa45f933680 _ZN4qucs3net13orderAnalysisEv (libqucs.so.0)
                                                         #2  0x00007fa45f933798 _ZN4qucs3net11runAnalysisERi (libqucs.so.0)
                                                         #3  0x000000000040290d main (qucsator)
                                                         #4  0x00007fa45f148f6a __libc_start_main (libc.so.6)
                                                         #5  0x0000000000402bea _start (qucsator)
```

and the .sch file is as follows:

```
<Qucs Schematic 0.0.19>
<Properties>
  <View=0,0,800,800,1,0,0>
  <Grid=10,10,1>
  <DataSet=dc.dat>
  <DataDisplay=dc.dpl>
  <OpenDisplay=1>
  <Script=dc.m>
  <RunScript=0>
  <showFrame=0>
  <FrameText0=Title>
  <FrameText1=Drawn By:>
  <FrameText2=Date:>
  <FrameText3=Revision:>
</Properties>
<Symbol>
</Symbol>
<Components>
  <GND * 1 100 250 0 0 0 0>
  <GND * 1 200 340 0 0 0 0>
  <Vdc V1 1 100 200 18 -26 0 1 "1 V" 1>
  <R R1 1 200 140 15 -26 0 1 "100 Ohm" 1 "26.85" 0 "0.0" 0 "0.0" 0 "26.85" 0 "european" 0>
  <R R2 1 200 240 15 -26 0 1 "50 Ohm" 1 "26.85" 0 "0.0" 0 "0.0" 0 "26.85" 0 "european" 0>
  <.DC DC1 1 80 40 0 37 0 0 "26.85" 0 "0.001" 0 "1 pA" 0 "1 uV" 0 "no" 0 "150" 0 "no" 0 "none" 0 "CroutLU" 0>
</Components>
<Wires>
  <100 230 100 250 "" 0 0 0 "">
  <100 110 100 170 "" 0 0 0 "">
  <100 110 200 110 "" 0 0 0 "">
  <200 170 200 210 "" 0 0 0 "">
  <200 270 200 340 "" 0 0 0 "">
  <200 170 200 170 "divide" 290 170 0 "">
</Wires>
<Diagrams>
</Diagrams>
<Paintings>
</Paintings>
```

in3otd, 2017-11-25:
um, @extrasharp, your test schematic here runs fine and produces the expected results.
I'll try to take a look around the functions mentioned in your backtrace to see if there is something suspect.

Are you able to recompile Qucs passing `--enable-debug` to `configure` and then use `gdb` to see exactly where `qucsator` is crashing?

extrasharp, 2017-11-25:
I was running off the binary from the Arch repos, however I cloned and built it myself.
Now I'm getting an error saying /usr/local/bin/qucsator can't be found?

I accidentally built it without --enable-debug so I'm rebuilding it now. Will update

in3otd, 2017-11-26:
do you happen to have a `QUCSDIR` environment variable set, from an old Qucs version maybe?
For my understanding, which repo did you clone, this one here (using the `develop` branch) or something else Arch-specific?