Qucs/qucs #719: Review changes on the netlist

guitorri, 2017-10-06:
This was caught with latest merges:

```
94m############################################�[0m
�[94m#  Report schematic to netlist conversion  #�[0m
�[91m--> Found differences (!)�[0m
{'AC_SW_resonance_prj': ['- Vac:V1 _net0 gnd U="1 V" f="1 GHz" Phase="0" Theta="0"\n',
                         '+ Vac:V1 _net0_TEMP_V1 gnd U="1 V" f="1 GHz" Phase="0" Theta="0" Ri="0 Ohm"\n',
                         '?             ++++++++                                          +++++++++++\n',
                         '+ R:_R__net0_TEMP_V1 _net0_TEMP_V1 _net0 R="0 Ohm" Temp = "26.85" Tc1 = "0.0" Tc2 = "0.0" Tnom = "26.85"\n',
                         '+ \n',
                         '+ \n',
                         '+ \n',
                         '+ \n',
                         '+ \n'],
```

A few things:
- Is it needed to output to internal resistance as a parameter of sources? A resistor is being added anyway.
- If so, we need to update the gold netlist.txt to reflect the internal resistance of the sources.
- Where are extra linebreaks coming from?

guitorri, 2017-10-06:
@felix-salfelder the *meters seem to be doing both, behaving as components (stamping the matrix), see d06558ce2e4ef, but also [hacking the netlist].(https://github.com/Qucs/qucs/blob/develop/qucs/qucs/components/component.cpp#L655-L657)

@gildias can you comment?

guitorri, 2017-10-07:
A few more notes.

Qucs:

phasordiagram, AC Phasorial diagram, <Phasor ... >
 - break backward compatibility, _Unknown diagram_
 - Diagram properties dialog, frequency should be a dropbox with available frequencies, not a textbox.
 - phasordiagram modify the <Diagram ... > signature, append the selected frequency https://github.com/andresmmera/qucs/blame/ImprovedMarkers/qucs/qucs/diagrams/diagram.cpp#L1406

Qucsator:

Silence warning:
```
In file included from ../../../src/components/component.h:43:
../../../src/circuit.h:136:29: warning: control reaches end of non-void function [-Wreturn-type]
  virtual int getstate () { }//only used in ohmmeter.cpp
```

andresmmera, 2017-10-07:
> Diagram properties dialog, frequency should be a dropbox with available frequencies, not a textbox.

I'm working on a fix for this...

andresmmera, 2017-10-07:
> break backward compatibility, Unknown diagram

Can we really solve this? I mean, these diagrams are new features not included in previous releases. I don't see how to tackle this issue properly.

I guess the same happens with the recently added CAPQ, INDQ and CIRCLINE components (I didn't check that...)

felix-salfelder, 2017-10-07:
> Can we really solve this?

once we have plugins, it will be possible to backport new stuff to old versions.

currently that does neither seem realistic nor feasible. right now we should think of proper exceptions and error messages to report missing parts (an unavoidable scanario).

guitorri, 2017-10-07:
Right now it is is impossible (or very hard) to expand the .sch syntax (add new stuff) without breaking backward compatibility. The current "parser" will just choke at any unknown thing. 
One way to handle it for now is add graceful degradation mechanism. When the user save the file with new stuff, inform that this version is not backward compatible. As a fallback, the `Save As` should enable to save for an earlier version. But since the .sch file format is not really versioned... tricky. 

(off-topic, but) This was one of the main reasons I cautiously opposed from merging the Qucs-S stuff. It is very useful stuff, but I fear that merging it unaware of the consequences it will frustrate users and push the burden of fixing stuff back to us. 

I guess is ok to break compatibility at some point (things have to evolve). For this reason people use  major, minor and patch releases to make it clear what is going on. We can argue that so far we only had patch releases, right? :smile: 

We should be fully aware of the consequences of breaking compatibility at all times, and work on our favor to help people move forward and minimize regressions and breakage on the user side. The main motivation for me on qucs-test was really this. To help us see the side effects, not to lock the things as is. We also need unit tests. I have something working and will push it shortly.

I am more concerned about the issue of internal resistance + additional resistance type of things (did not yet check if that is indeed a problem).

Enough of my gibberish. 

As of now it seems that the above PR needs a bit more work. I am also looking into it.

guitorri, 2017-10-16:
Updated the issue description and action list.

My question to all is the following: Shall we enable internal resistances for iprobe, vprobe and wprobe?

I am in favor of keeping them ideal. Users can add resistors as needed to make the components more realistic.

felix-salfelder, 2017-10-16:
in the light of your updated to-do list ("nobody has implemented that anyway"), i suggest to stick to old/legacy probes.

newer code will have no problem with users bringing their own probes, if they like.

andresmmera, 2017-10-16:
I'm also in favor of keeping them ideal. I admit I accepted and merged #682 but, honestly I think that the sources should be an ideal entity...

guitorri, 2017-10-16:
Don't worry @andresmmera. I will push the fixes shortly.

guitorri, 2017-10-17:
I will remove the ohmmeter. 

The way it is implemented messes with the nasolver, trsolver, acsolver, dcsolver and the circuit baseclass just to get the information it needs. I don't think this is right approach. 
I am unsure about the uses of a dedicated ohmmeter in simulation. If the user need to know the impedance, open the branch he/she wants to inspect, add a source an probe in and that is it.

This was clearly targeted as an educational tool. The first failure mode is partially covered, if there are other sources actives the ohmmeter gives a NAN reading (which does not help much). The second mode can be confusing for beginners, if there are more than one ohmmeter on the same circuit it gives wrong results.

gildias, 2017-10-23:
Hello,

With all the added modules on #682 being targeted towards engineering/electronics students, one of the main objectives was to simplifiy the usage of the simulator, as such, the ability to choose the desired frequencies for the phasor diagram is a great improvement, keeping this philosophy in mind. To clarify, in its current state, the Wave(Temporal) diagram does not display multiple frequencies at once. 

The netlist was altered so a resistance was added in parallel to the necessary components, with the example of the ammeter, to mimic a "real" component. In the "real" components in which the internal resistance would result in a series with its other parameters, it was directly introduced to those components. 
Obviously this can also be achieved by adding another resistor to the circuit, but the addition of this function, even if a different approach is followed, will result in a more complete simulator.

The ohmmeter was implemented in a way that it could reach the same values as the voltmeter, as it needs that information to calculate the impedance in the branch, furthermore it should only cause interference whenever it is detected in the circuit. 
As you mentioned, if more than one ohmmeter exist in the same circuit, it will give a wrong result, this is probably because they act as separate current sources, and interfere with the values of one another, a possible way to solve this issue would be to return some warning when more than one ohmmeter is connected to the same circuit. Did not find a way to do this while still enabling multiple ohmmeters per simulation ( as multiple circuits are supported and functional ).

guitorri, 2017-10-23:
Take a look at iprobe, vprobe and wprobe. The implementations for these probes are not leaking out of their respective `.cpp` files.  
The ohmmeter is spreading implementation and logic into several simulation modes. Not to mention that it traverses the circuit at each time to find the `CIR_OHMMETER`. It does not seem right and it is going to be maintenance burden. 
It is almost like the ohmmeter requires a dedicated simulation mode of its own, to take care of disabling additional sources and checking for multiple definitions of ohmmeters on the same circuit.

Perhaps students should be taught to drive a unity voltage source and measure the current. One day they will be confronted with real circuits or with another simulator and they will need to know that much.

felix-salfelder, 2017-11-07:
done. thanks

closed 2017-11-07 by felix-salfelder