Qucs/qucs #734: Fix fmod recursion

guitorri, 2017-10-13:
```
git/qucs/qucs-core/src/math/real.cpp:256:36: warning: all paths through this function will call itself [-Winfinite-recursion]
nr_double_t fmod( nr_double_t arg) {
```

Remove the `qucs::fmod` ? Use `std::fmod` direclty?

in3otd, 2017-10-13:
um, it seems that since some time we use `std::fmod()` [anyway](https://github.com/Qucs/qucs/commit/8903260efa78f90e7b71b4e866d542c120943065#diff-a00a113d45f536de2592768556b7c8d9L914); didn't see where else it's used.

BTW, [checking](https://github.com/Qucs/qucs/blob/develop/qucs-core/src/math/real.cpp#L257) for `HAVE_STD_TRUNC` in the `fmod()` routines seems weird :grin:

guitorri, 2017-10-13:
In fact, I think the only place fmod is used is below:
```./src/evaluate.cpp:974:  _RETD (std::fmod (d1, d2));```

Guess who is to blame for most of those tests? :smile: