Qucs/qucs #745: docbuild fails

felix-salfelder, 2017-10-18:
`../configure; make` ends here

    ! Class scrreprt Error: undefined old font command `\tt'.
    
    See the scrreprt class documentation for explanation.
    Type  H <return>  for immediate help.
     ...                                              
                                                      
    l.56 found in the file {\tt
                                BFP450\_3V040M.S2P}, available from the
    ?

just for the record. haven't tried to debug yet.

ralvgh, 2017-11-22:
LaTeX help says ...

> ? You should note that since 1994 LaTeX2e provides a new font selection scheme
called NFSS2 with several new, combinable font commands. KOMA-Script
classes had defined the old font commands like `\tt' only for compatibility
with old LaTeX 2.09 document styles of Script 2.0. Nevertheless, these
commands are deprecated and undocumented at least since 2003. Since 2013
KOMA-Script classes warned about soon removement of these deprecated commands.
Now, after two decades of LaTeX2e and NFSS2, these commands will not work any
more. If loading a package results in this error message, you should contact
the author of that package and ask him to replace the deprecated font command
`\tt', e.g., by `\normalfont \ttfamily `. Otherwise you should reconfigure
or replace the package. If you have used the old font command `\tt' yourself,
you should replace it, e.g., by `\normalfont \ttfamily '.

... and doing as told (see [patch.qucs.issue.745_2017-11-22-1736.txt](https://github.com/Qucs/qucs/files/1497023/patch.qucs.issue.745_2017-11-22-1736.txt)) worked fine.

in3otd, 2017-11-22:
thanks; could we use also `\texttt{}`, like done in several other places in the docs?

ralvgh, 2017-11-22:
(... after digging and verifying long forgotten LaTeX ...) @in3otd, you are right, we **should** use `\texttt{}` given the way `\tt` is used in the patched file (as inline formatting command and not as setting for eg a big block of text); apologies for the quick-and-dirty "doing-as-told" ...
[patch.qucs.issue.745_2017-11-22.2321.txt](https://github.com/Qucs/qucs/files/1497714/patch.qucs.issue.745_2017-11-22.2321.txt)

felix-salfelder, 2017-11-23:
i have imported the updated patch. see #752

in3otd, 2017-11-25:
thanks, merged in 319e8ab39f688333340fe09efa00fe8995b3764c

closed 2017-11-25 by in3otd