Qucs/qucs #746: Select independent variable to draw diagrams with if some variable is dependent on more than one.

kenogo, 2017-10-28:
I want to run a simulation on a frequency dependent voltage divider. I use an AC simulation to look at the output voltage in dependency of the frequency. Now, I also want to run a parameter sweep for the resistor's resistance. When drawing diagrams, is there a way to draw the voltage in dependency of the resistance for one arbitrary frequency (say f=1MHz)?

Thanks for this awesome project by the way! :)

in3otd, 2017-10-28:
maybe you could do something with `yvalue()` and `PlotVs()` (see the docs [here](http://qucs.sourceforge.net/docs/tutorial/functions.pdf))

felix-salfelder, 2017-10-28:
found an example showing repeated dc simulations while changing a parameter.
https://github.com/Qucs/qucs/wiki/Parameter-Sweep#Sweeping_a_design_variable

does this not work with ac?

kenogo, 2017-10-28:
Thanks. I eventually managed to do it using array slicing after figuring out how the arrays are ordered. `PlotVs()` looks promising though.

kenogo, 2017-10-28:
I did something like
```
voltage=output.v[0:length(acfreq),x]
```
Where `acfreq[x]` is the arbitrary value for `acfreq` that you're getting the values from.

closed 2017-11-12 by kenogo