Qucs/qucs #753: To remove the Qt3 supporting classes

eduard-x, 2017-11-29:
Hello!
I'm testing now the Qucs projects without supporting Qt3 classes. I't possible to compile with Qt4 or Qt5.
For testing i need so about 1-2 weeks, after this i can push the patch. 
Best regards,
Eduard

[image]

in3otd, 2017-12-01:
uh, you mean you removed all the Qt3 support classes usage from Qucs? Quite a nice work!
AFIU, @guitorri was also working on this (and more), would have been a good thing to coordinate the work.
Looking forward to test the code :grin:

guitorri, 2017-12-04:
Please clarify, did you just remove the legacy code to make the compiler happy or did you also ported the functionality forward? I mean, does the UI still works?

Also, from the title bar it seems that you are based on `qucs-s` which is currently a fork from `qucs`.

Yes I did some work, but I am currently stuck around the mouse event handling code. I am trying to find the path of less friction (minimal changes to current implementation) to get the UI to work with the QGraphics framework.

It is far from finished, but I will post the code anyway in a few days.

eduard-x, 2017-12-04:
Hello!
Of course, not only compiling :) Functionality is in checking now. Compiler was happy as i made the screenshot. Now i work with fork qucs-s, but the maintainer of this is now very busy, and i can help yours project. :)
Best regards,
Eduard

eduard-x, 2017-12-14:
Hi!
I transfered the some changes from Qucs-S project. This now is testing of QGraphicsView() based visualization in Qt4. On this week i have to test the mouse actions. I'm ready to push step by step the changes in separate branch. 
Regards,
Eduard
[image]

guitorri, 2018-01-03:
@eduard-x, please share your progress so we can work together.

I pushed the things I have done so far to https://github.com/guitorri/qucs/tree/QG-port-04

I still don't know how to handle the mouse actions. I was thinking about keeping the `MouseActions::` functions signatures which are passed around as pointers and modified as the UI changes state. The thing is that some `MouseActions::` functions are doing the updates on the graphic item's member variables. I think the items should update themselves...

eduard-x, 2018-01-06:
Hi! Can add the branch in your fork and we can test it together. I left the all MouseActions functions, but the code need small changes for scrolling, reactions from vertical/horizontal scrollbars. The other big change is the removing of Q3PtrList, my solution is the PointerList class, based on QList. 
Happy New Year! :)

guitorri, 2018-01-08:
@eduard-x, I added you as collaborator to my fork. You should be able to push. Best wishes! 😄

eduard-x, 2018-01-10:
@guitorri ,  i'll push the source code step-by-step, all what i have, not all files in one evening :)
best regards

eduard-x, 2018-01-15:
The source code is pushed in wip/eduard/removing-qt3 branch, qt3 is deactivated. Now i test the project with Qt5 compiling.

guitorri, 2018-01-15:
Thank you. I pushed a small fix. I will keep testing with Qt4. At least until we get things working, merged and released. I am waiting to see what else you have done so we can discuss what still needs to be done. I kept a list of thinks I broke or left to be done later.

eduard-x, 2018-01-16:
The fixes for Qt5 compiling and source code compatibility fixes are pushed. :)
The project was compiled on Linux system, possible Win/Mac CMakeLists mistakes.

a17r, 2018-03-18:
What's the current state of this? Can we get an estimate for a possible Qt5 release?

eduard-x, 2018-03-18:
@a17r 
the source code was pushed for compiling with Qt5 on 16 jan 2018 https://github.com/guitorri/qucs
in branch wip/eduard/removing-qt3

guitorri, 2018-03-18:
No estimate. edouard-x added a compatibly layer to replace the Q3PtrList class (not sure if it complete or fully tested). The remaining issue is  the canvas that need to be ported to the QGraphics framework. Things are starting to work, as in https://github.com/guitorri/qucs/tree/QG-port-08. I will keep pushing incremental branches until things stabilise.

closed 2018-03-19 by eduard-x