Qucs/qucs #754: Package Qucs in flatpak?

Juankz, 2017-11-30:
Many people struggle installing Qucs and also building from source is quite hard (Everybody uses Qt5 nowadays but Qucs is using Qt3/4 and some OS already dispose those libraries). So I have seen that [flatpak](https://flatpak.org/) is an easy way to distrute Qucs and get rid of those problems. Is it possible add it to the roadmap

felix-salfelder, 2017-11-30:
Thanks for your interest in qucs.

what changes do you need in qucs in order to distribute it with flatpak?

Juankz, 2017-12-01:
Well I haven't dig much into flatpak or Qucs, but it looks like you only need a manifest.json file following [flatpak starter tutorial](https://flatpak.org/hello-world). More detailed information can be found [here](http://docs.flatpak.org/en/latest/)

felix-salfelder, 2017-12-01:
what's the difference to other package ecosystems, such as conda or to operating systems such as debian or app platforms such as f-droid?

i do not understand: why do *we* need a manifest rather than them?

Juankz, 2017-12-01:
Well flatpak has many advantages over traditional distribution systems.

Flatpak is universal packaging for GNU/Linux distros, that means you don't need to worry about packaging for debian, ubuntu, fedora, arch, etc. Just one package that runs in all of them. 

Flatpak also has the advantage of faster updates, as soon as there is a new release, people will use it, they would not need to wait for Debian to update it (usually Debian packages updates are very slow)

Another advantage, and this is my use case, I run Solus operating system which does not support Qt3 nor Qt4 libraries, only Qt5, therefore compile and run Qucs right now is impossible. As flatpack support sandboxing, Qucs will get those libraries independently from the OS and problem solved.

What do you think?

felix-salfelder, 2017-12-01:
> don't need to worry about packaging for debian

we don't do that. debian folks worry about packageing for debian. flatpak will not get in their way, as it likely has other goals.

> get those libraries independently

which i think is a terrible idea. just pulls in more work that others (can) do better.

> What do you think?

packageing qucs for anything is definietely a right thing to do. so unless you use your time for fixing the qt3 issue ... please go for it.

in3otd, 2017-12-01:
not sure I understand flatpak, but if you have a script or something to build a package for that we could add that to the [/contrib](https://github.com/Qucs/qucs/tree/develop/contrib) directory

Juankz, 2017-12-01:
Ok copy that

closed 2018-05-10 by guitorri

Murmele, 2021-01-16:
I created one, based on cmake and the qt5 branch. It is working, but the qt5 branch is missing still a lot so it is not usable. But you will get it