Qucs/qucs #759: Simple touchstone export

arsenovic, 2017-12-11:
how do i do a simple touchstone export? 

i can not figure out how to use the import/export gui.

in3otd, 2017-12-11:
yep, it's a bit cumbersome and IIRC not properly described anywhere.

Select Project->Import/Export Data (note that you *must* open a project, for some unknown reason)
then select the desired Qucs result file (`.dat`) as Input File, type an Output File name, select Touchstone in the drop-down menu on the Output Data line. You may need to type **S** there, to mean you want to export the data from the `S` variable but if that's the only variable in the dataset it may not be needed. Click on the Convert button and check for no errors in the Messages window.

You can do the same probably faster from the command line:
`qucsconv -if qucsdata -of touchstone -i results.dat -o results.s2p`

arsenovic, 2017-12-11:
ahhh i see. the dat is in the project folder.  thanks a lot . 

i was trying to export a dat from the display sheet and got a 'select a diagram error'. 

a simple s-parameter simulation would make a nice doc.

closed 2017-12-11 by arsenovic

cweickhmann, 2020-06-17:
Since I found this, I have a few observations though:
* The dialogue + executed script does not seem to be able to handle spaces in folder/filenames
* The output file seems to be w.r.t. the input file's directory? Or the project folder? This is unclear and cannot be specified.
* The error messages are sparse to say the least. Is the "wrong file format" because of the input file or the output file?
* Since the .dat file contains a finite list of properties, couldn't they be shown as a list under output data?
* After one Error, the "Convert" button stays deactivated. So one mistake and you have to re-open the dialogue :-(

I am not a developer, so I currently have nothing to offer to fix this but this list of imho obvious, urgent fixes.