Qucs/qucs #761: Misbehavior of the color picker button in Windows 10

andresmmera, 2017-12-19:
I've recently installed the [latest Qucs (develop branch) Windows binary](https://ci.appveyor.com/project/qucs/qucs/build/1.0.1533/job/9ftu4qn4uilu4brx/artifacts) in a Windows 10 machine and I've noticed that the button which pops up the color picker only works when the mouse pointer is in the gray area.

Unfortunately, I couldn't reproduce this here in Ubuntu and I cannot do Qucs development work in such Windows box. Can anyone confirm this bug and (hopefully) take a look to it? I guess it comes from #672 

[image]

andresmmera, 2017-12-22:
> I think this may be just one more of those strange Qt issues...

Honestly, I had the same impression... So it seems that we can't do anything with this, right?

guitorri, 2018-07-25:
I guess this is a won't fix kind of bug. I really think it is a but in Qt. I tried poking around a while ago, but could not find a satisfactory solution without subclassing QButton and more... I don't want to do that.

closed 2018-07-25 by guitorri