Qucs/qucs #765: Snap support

BearzRobotics, 2018-01-01:
On your website I see that the team officially supports Ubuntu with a ppa, I was wondering if you guys were planning on also making it snap. The snap has the advantage of not having to add a pp, plus its install able on other distros like Solus, Debian, arch, etc (Any that support snapd). Plus Launchpad has support for creating snaps on multiple architectures from ppas.

eldarkg, 2018-04-22:
@BearzRobotics You can see src of snap package here https://github.com/eldarkg/qucs-spice-snap.
This package contains qucs-spice branch.

guitorri, 2018-05-10:
I guess someone (developer or not) has to step up and maintain this packaging system. 
If there is something that needs to be added to our repository, by no means, let us know and we will put in under contrib. Closing for now.

closed 2018-05-10 by guitorri