Qucs/qucs #768: icons not showing

felix-salfelder, 2018-01-07:
found in #742, but unclear how it relates to the changes there.

back to `develop`, i do

    $ cd qucs
    $ ./bootstrap
    $ mkdir b; cd b
    $ ../configure; make
    $ qucs/qucs

and i (still) see no icons in the gui. maybe it's just me, but i need a fix before continuing with #742...

in3otd, 2018-01-08:
FWIW, following the steps above I do see all the icons, as usual (tried also on a fresh, clean, clone of `develop`, no Qucs installed anywhere, still have the icons)

felix-salfelder, 2018-01-08:
thank you!

looking for further differences, i finally found it.

    $ rcc -version
    rcc 5.9.1
    $ export QT_SELECT=4
    $ rcc -version
    Qt Resource Compiler version 4.8.7
    $ rm qucs/qucs_.cpp
    $ make
    [..]
    $ qucs/qucs

now i see icons again. somewhere i read that [Qt resources] must be linked into the main executable [for now]. perhaps they fixed that, but not in a backwards compatible way...

in3otd, 2018-01-09:
interesting... your distribution then does not provide a `rcc-qt4` by default, which I thought was fairly common when also another Qt version is installed. AFAIR, `configure` looks first for that and if not found it looks for just `rcc`.
(FYI, here I have `rcc` (Qt4 version) and `rcc-qt5`)

felix-salfelder, 2018-01-09:
the qt5 resource file uses a different binary layout, but also tells qucs about it. the first argument to `qRegisterResourceData` is `0x2`. it is `0x1` in what rcc-4 creates (on my system).

unfortunately, qt4 does not even emit a warning if that version number is too high.

NB: my shiny new rcc also has a `--format-version <number>` flag. (no, i am not convinced, i'd prefer not to burn bridges.)

PR #807 (2018-07-11): Qt selection wip
this is going to fix #768.

... and necessary for the Qt4/5 rework
closed 2018-07-11 by guitorri