Qucs/qucs #771: ACSimulation sometimes does not work depending on start freq ASCII

ScienceNut105, 2018-01-26:
Sometimes ACSimulation does not run but it can be fixed by changing how start frequency is entered.
Try "1 KHz", "1K Hz", "1 K Hz" and ACSimulation will start working.  0.0.19 and windows 10.

ScienceNut105, 2018-01-26:
Happens on example of page 30 QUCS a tutorial getting started with QUCS Jahn and Borras.

ScienceNut105, 2018-01-26:
Better copy of netlist
# Qucs 0.0.19  C:/QUCS_PROJECTS/RC_HiPass_prj/RC_HiPass.sch

C:C1 gnd Vout C="1 pF" V=""
Vac:V1 Vin gnd U="1 V" f="1 GHz" Phase="0" Theta="0"
R:R1 Vout Vin R="50 Ohm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
.AC:AC1 Type="lin" Start="1 Hz" Stop="1 KHz" Points="101" Noise="no"
last line is grey and blank.

ScienceNut105, 2018-01-26:
repopen

in3otd, 2018-01-26:
please attach the `.sch` schematic file and the `netlist.txt` file created by Qucs in the "Qucs Home" directory (should be `C:/QUCS_PROJECTS` from the text pasted above)

ScienceNut105, 2018-02-03:
Example pg 33-35 RC_HiPass in A tutorial Getting Started.  If you type KHz for stop says expecting "".
If you type Khz it makes frequency 0 to 1 Hz.  If you go to small k and type "1 k", it goes 0 to 1000 Hz.
Appears to not like Capital K.

andresmmera, 2018-02-03:
Honestly, I couldn't take a look at this issue, but could it be somehow related to #409 ?

in3otd, 2018-02-03:
> Appears to not like Capital K.

Yes, Qucs does not accept **K** (uppercase) as unit prefix since the correct prefix to indicate a factor of 1000 is **k** (lowercase), see e.g. https://en.wikipedia.org/wiki/Kilo-

closed 2018-07-19 by guitorri