Qucs/qucs #772: configure flag inconsistency: mkadms vs. admsxml

couchand, 2018-02-07:
It looks like the changes in c6e7b03eab4a115720da59460ee1289ec9945027 didn't quite get completed.  Though the readme suggests to use `--with-admsxml` to specify an alternate admsXml location, it looks like the flag is still `--with-mkadms`.  Am I missing something obvious here?

felix-salfelder, 2018-02-07:
perhaps it should be `--with-admsXml`. that corresponds to what it does now. i don't know what `mkadms` was supposed to be.

NB: qucs (the gui) should not depend on anything adms at all. there's some cleanup required still.

felix-salfelder, 2024-07-03:
No more adms involved in the gui configuration process. Not sure if this is all about qucsator, please reopen there if needed.

closed 2024-07-03 by felix-salfelder