Qucs/qucs #773: can't build on gentoo (troubles with latex doc probably)

rpolasek, 2018-02-07:
[qucs_build.log](https://github.com/Qucs/qucs/files/1702334/qucs_build.log)

in3otd, 2018-02-08:
it seems that the `IEEEtran.bst` BibTeX style file is missing; I'm not familiar with Gentoo and I don't know if it has a specific package for that.
Anyway, an easier solution will be to pass `--disable-doc` to `configure` to avoid (trying) to compile the docs. Ready-made PDFs for the docs can be found at https://sourceforge.net/projects/qucs/files/qucs-binary/0.0.19/

rpolasek, 2018-02-09:
thanks for your support

closed 2018-02-11 by guitorri

in3otd, 2018-02-15:
I think we should revert to having the Latex docs **not** built by default, at least until all the proper checks in `configure` are in place, since "not able to build the docs" is becoming the most reported issue (latest, as of today, [here](https://sourceforge.net/p/qucs/support-requests/11/))