Qucs/qucs #780: optional things

felix-salfelder, 2018-02-15:
there are issues with optional things that are entangled with the main project. one is the documentation, the other is the cmake build system. there may be more in the future.

i propose to adopt the "anything that *can* be optional, *must* be optional" paradigm. this will improve maintainabilty, and enable progress. it will also make contributions possible at a later stage.

c.f. #773, #757, #742

guitorri, 2018-02-15:
Cmake build is not on par with Autotools (yet), but it is certainly not optional. 
It significantly lowers the barrier for new contributors that are familiar with IDE environments such as QtCreator. 

For the rest, documentation, adms, ..., I am ok with making them optional. As long as we keep one integration branch/repository to make sure things still work together.

felix-salfelder, 2018-02-15:
> It significantly lowers the barrier for new contributors

(maybe) true. currently it mainly makes refactoring impossible. refactoring requires work on the plugin system, which somebody needs to implement in a way that works on all platforms, plus incremental changes to make sure it does not break. somebody does not know how to deal with cmake.

in the far end, contributions will be possible as plugins, plugins can bring their own build systems (even scons, if you'd like).

are we in a deadlock?

i have no problem with cmake, i just can't maintain it. i am thinking of a model similar to gnucap. gnucap provides an autotools build system, which does *not* interfere with upstream development, while anybody can still use it, at will. (what's wrong with that?)

felix-salfelder, 2020-09-26:
cmake will be dealt with in #1007. other stuff will become optional during the refactoring, either way,

closed 2020-09-26 by felix-salfelder