Qucs/qucs #781: How the simulation works in Digital?

Kiddinglife, 2018-02-16:
Hi there,
Can you tell me how the simulation of digital circuits  works consisted of the devices defined in the folder of qucs/qucs-core/src/components/digital?
Actually, I am comparing qucs with logicsim[](https://github.com/hneemann/Digital).

guitorri, 2018-02-17:
The built in digital simulation models in qucs-core are not event based.

The components are single macromodels for time domain simulation. They allow for very simple mixed-signal (analog and digital) simulation. 
Search for "logical" or "boolean devices" on the Technical Manual for details on the transfer characteristics of the logical functions.

Qucs-Schematic can drive digital simulation by converting the schematic to drive HDL code on either Icarus-Verilog or FreeHDL engines. These are event based engines. 

GHDL, Verilator (... and other) tools also implement event based simulation but are not integrated in to Qucs.

Gnucap can handle mixed signals (time domain and events).
Ngspice with XSpice extension can also do mixed signals.

Kiddinglife, 2018-02-20:
thank you so much! I can close this issue now.

closed 2018-02-20 by Kiddinglife