Qucs/qucs #783: .sch file associations?

kf4mot, 2018-02-22:
With 0.0.19 and on win7 x64 and win10 x64 when trying to open a example when you already have software that has associations with .sch files (Pads by Mentor) just trys to open the file with the software that has that association.

in3otd, 2018-02-22:
I assume you mean this happens when choosing File -> Examples ; this just opens a standard File Explorer, so the default application associated with `.sch` files will be used to open them - Qucs does not change the associated application.
If you want to browse the Examples, currently the best way is to copy all them to the Qucs home directory and open them using Qucs. There is currently some work ongoing (#763) to make this more straightforward but it's not finished yet.

kf4mot, 2018-02-22:
That works. Thanks!

closed 2018-02-22 by kf4mot