Qucs/qucs #788: Error Occured When Compiling qucs-0.0.19

Kathy626, 2018-03-22:
Hey, all
I am trying to compile qucs on Linux machine, but there are several error popped out when I was trying it.  First, I got three errors during make:
make[3]: *** [gperfappgen] Error 63
make[3]: Leaving directories `/home/user/documents/qucs-0.0.19/qucs-core/src'
make[2]: ***[all-recursive] Error 1
make[2]: Leaving directories `/home/user/documents/qucs-0.0.19/qucs-core/'
make[1]:***[all] Error 2
make[1]: Leaving directories `/home/user/documents/qucs-0.0.19/qucs-core/''
make: ***[all-recursive] Error 1z

Can anyone give me a hand of what happened?I have been stuck for compiling QUCS for several days already

in3otd, 2018-03-22:
could you please post
- the output of the command `gperf -v`
- the `config.log` that you have in `/home/user/documents/qucs-0.0.19/qucs-core/`
- all the messages you get after going to  `/home/user/documents/qucs-0.0.19/qucs-core/src` and typing `make gperfappgen`

Kathy626, 2018-03-22:
Thanks for replying
I think I found the reason, somehow, qucs has two version of libtool installed under qucs.
one is 2.4.6 and the other one is 2.4.3. I checked the original qucs package, there is only one libtool (version 2.4.3). But after ./bootstrap (or some other initial installation), another libtool has been added into qucs.
have you ever met this situation before?

closed 2018-07-19 by guitorri