Qucs/qucs #789: S-parameters parser fails on big files

in3otd, 2018-03-22:
S-parameters file parser fails on (relatively) big files, see e.g. https://sourceforge.net/p/qucs/discussion/311049/thread/4aeb8b6e/ . I can reproduce this on a 100 MB file, with about 600000 lines.
Increasing YYMAXDEPTH as suggested can solve the issue but looking at the Bison docs it seems that [defining rules differently](https://www.gnu.org/software/bison/manual/html_node/Recursion.html) may help.

Kathy626, 2018-03-22:
I have met the same issue as you did also, YYMAXDEPTH is one reason caused this issue and left/right recursion is the other reason leads YYMAXDEPTH only up to one limits.
this memory size limitation is in qucs, qucsator, qucsconv.

felix-salfelder, 2018-03-23:
maybe it's easier to do it with the stream tokenizer shipped with stl.
roughly

    std::vector<sth> mynumbers;
    std::file f(filename);
    somehow_parse_the_header_here(f);
    while(!f.eof()){
          double d;
          f >> d;
          mynumbers.push_back(d);
    }

excuse my ignorance -- i can't see any interesting grammar here that would justify the (mis)use of lex and yacc.

PR #791 (2018-05-10): Use left recursion for parsing S-parameters files
Fix #789.
Seems to work fine on all the S parameters file I have at hand here (from 2 to 40 ports and size up to 100+ MB).
closed 2018-05-10 by guitorri