Qucs/qucs #790: qucsconv is not working under Linux

Kathy626, 2018-03-22:
Hey, all
I have met a problem relates to qucsconv in Linux.
I've tried to use qucscpnv to convert a touchstone file (s38p) into qucsdata file (dat), this command works totally fine in my Windows machine, but it errored out in Linux, and I have attached error message in here:
[image]
Does anybody has met this problem before?

Kathy626, 2018-03-27:
I've found out that since the touchstone file I am using is s38p file, which for each frequency point is 38 x 38 matrix. Not sure whether Linux qucs has the ability to handle this big file

in3otd, 2018-03-27:
these errors come from the routine that parses the S-parameters file; it may be somehow related to #789.
I'm currently doing some simulations with a 36-port S-parameters file and it works fine in Linux - could be faster probably but the results are ok.
The problem *might* be related to the line terminations, you could try to change that using `dos2unix` or similar and see if it makes a difference. 
It will of course be useful if you could share a stripped-down/reduced version of the problematic file.

Kathy626, 2018-03-27:
thanks a lot, very very helpful

Kathy626, 2018-03-27:
Which file you'd like me to share? the reduced touchstone file or compiled code?

in3otd, 2018-03-27:
I'd like to have a Touchstone file which you are not able to convert on Linux so I can try to use that to debug this issue locally.

Kathy626, 2018-03-28:
The file is too big to upload, even after I zip the file still too big.
The touchstone file I have is 38 ports (s38p), with 1315 frequency sweep points.

Kathy626, 2018-03-28:
​
 Test_sparams_38ports.zip
<https://drive.google.com/file/d/1Z09_rcd2lg_VXzbbrOINgkgxuYM_5v9g/view?usp=drive_web>
​does this work?

in3otd, 2018-03-29:
thanks for the file; here I'm able to convert it without errors both in Linux and Windows. I've run a quick simulation, the results seem a bit strange (magnitude often >> 1 and not smooth) but I guess this may be ok. Not sure where the problem could be there; you may try to reduce your file to just 1 or 2 frequency points and see if it works fine.

Kathy626, 2018-03-29:
> yep, that's strange indeed. I saw errors similar to what you reported only
> recently when playing with the Bison grammar file to fix #789
> <https://github.com/Qucs/qucs/issues/789> but never with the default
> grammar file. Here in Linux I'm using bison (GNU Bison) 3.0.4 with flex
> 2.6.0.

Ok, maybe the version is the reason causing this strange thing. Because,
from my downloaded QUCS Linux package, Bison is 2.7, flex is 2.5.37

But I am so confused of how you could observe the similar error when you
compiling QUCS with Bison Grammar, this should not affect

Kathy626, 2018-03-29:
Thanks. By the way, do you know how to change Bison version for qucs?

in3otd, 2018-04-05:
`bison` is using only when compiling Qucs to build the parser source code; if you update your Bison version you need to recompile Qucs to see if the new version works differently. How to update Bison depends on the Linux distribution.

closed 2018-07-19 by guitorri