Qucs/qucs #792: Wrong cutoff frequency for Chebyshev and Elliptic

thomaslepoix, 2018-04-20:
I found out that cutoff frequency is around -3dB for every type of filter. But convention says we should use ripple amount for filters that exhibit ripple (Chebyshev, Elliptic).

[image]

in3otd, 2018-04-20:
it's true that many filter design tool use the ripple amount to define the cutoff frequency of filters with ripple but personally I find this rather annoying - I do not see why the definition of "corner frequency" should change depending on the type of filter.
In `qucs-filter` the Chebyshev filters corner frequency is assumed to be the - 3 dB frequency but this seems not to be the case for the Cauer filters. So in the end it's certainly confusing, we should either always use one definition for the corner frequency or clearly state which definition we are using.

thomaslepoix, 2018-04-21:
For me Cauer is -3dB defined too.

I agree with you but the convention is what it is, so it might be right to let the user choose his cutoff frequency definition, don't you think?

in3otd, 2018-04-21:
do you have a design example where the Cauer cutoff frequency appears to be at -3 dB instead at the ripple amount? I did a few tests here and it really seems to me that the -3 dB frequency is used. I will need to check the code but this will take some time.

I think that we should make it clear what definition is being used at first, if then we can let the user choose that's of course even better.

thomaslepoix, 2018-04-21:
Oh sorry you're right, I once made this with 5° order filters and Cauer seemed to be -3dB designed.
[image]

But I just checked 3° order filters and it is obviously not.
[image]