Qucs/qucs #795: Appimage package support

eldarkg, 2018-04-22:


guitorri, 2018-05-10:
I guess one has to step up and maintain this packaging system. If there is something that needs to be added to our repository, by no means, let us know and we will put in under contrib. Closing for now.

closed 2018-05-10 by guitorri

nuess0r, 2018-06-12:
Please reopen this issue and search for a current developer to produce a working appimage of Qucs 0.0.19.

Let me explain why an Appimage could be important for the future of Qucs and attract contributors:

First, Appimage is not another package system like DEB. It's more like a DMG on OSX or a portable application under Windows. 

> The key idea of the AppImage format is one app = one file. Every AppImage contains an app and all the files the app needs to run. In other words, each AppImage has no dependencies other than what is included in the targeted base operating system(s). https://appimage.org/

In my opinion this is very important for Qucs to keep it running under Linux. At the moment it is virtually impossible to install Qucs on a recent Linux distribution because of the QT3 library dependencies. Qucs silently disappears from all major Linux distributions because of this and you are loosing a (my quess) larger part of your user base.

Qucs is great, gives reliable simulation results and with its good UI it is a good candidate to use in education (Not only universities but especially teenagers already knowing Arduino stuff, technical colleques and high school courses, HAM radio enthusiasts etc.) and I would hate loosing it.

Why I don't do it myself: I do not know my self how to get Qucs compile and where to get all the dependencies for QT3. That's why I kindly as an active developer to provide an Appimage so we Linux users can spread the word about Qucs and hopefully attract more contributors.

felix-salfelder, 2018-06-12:
On Tue, Jun 12, 2018 at 03:10:26AM -0700, nuess0r wrote:
> Why I don't do it myself: I do not know my self how to get Qucs
> compile and where to get all the dependencies for QT3.

Sorry for an off topic remark. I was under the impression that the
current development version ("develop" branch on github) builds and runs
with qt4, in principle.

Please retry and report any specific issues we do not know about yet.

nuess0r, 2018-06-12:
Hi Felix

Thanks for your mail.

> On Tue, Jun 12, 2018 at 03:10:26AM -0700, nuess0r wrote:
>
> Sorry for an off topic remark. I was under the impression that the
> current development version ("develop" branch on github) builds and
> runs [...]

I guess you are right. I don't know better because I still use Qucs
0.0.15. This version I installed ages ago and since then the Ubuntu
packages never worked on my Debian Sid.

I never tried it to compile my self because it is often very frustrating
when you are using Debian Sid like me (especially for large projects).
Sid works very well with packaged stuff but often you use newer tools
than the original developer and you run into "funny" problems.

I would like to help out as package maintainer (there are other
for me relevant packages that would need someone to adopt them) but I
never did something before and also I have no time left to this. 
(I'm a founding member of a hackerspace and the publicdomainproject.org
is consuming all my free time)
 
> Please retry and report any specific issues we do not know about yet.

I still believe that an Appimage would increase the Linux user base
again and removes pressure to find package maintainers.

All the best,
nuess0r