Qucs/qucs #796: State of the resistor color codes tool

crazyfermions, 2018-05-06:
The *resistor color codes* tool has several issues:

- [ ] Bug: Resistors cannot be pasted into the qucs schematic (wrong format error)
- [ ] Bug: Wrong/duplicate entries of resistor colours for the forth column (supposedly the multiplier)
- [ ] Enhancement: GUI offers no explanation what the colour codes mean which would be nice for non-expert users, maybe add an [explanation](https://www.digikey.de/-/media/Images/Marketing/Resources/Calculators/resistor-color-chart.png?la=en-DE&ts=03de8cd9-c3fe-4182-ae7d-4d0decc83ad9) or at least add names to the colour columns
- [ ] Enhancement: Overall refactoring of code, e.g. excessive use of (class) names like *mywidget.cpp*

andresmmera, 2018-05-16:
I can reproduce that. 

To be honest, I'm not too familiar with the resistor color code tool, but I've been taking a quick look at the source... 

I noticed that it takes the number of bands from the tolerance combo (see [this](https://github.com/Qucs/qucs/blob/develop/qucs/qucs-rescodes/resistor.cpp#L118))
Is that correct? I believe the number of bands is not related to the resistor tolerance...

https://en.wikipedia.org/wiki/Electronic_color_code#Resistor_color-coding

in3otd, 2018-05-16:
I think that the code is right, in practice: for resistor with up to 5% tolerance you need just two digits (plus a multiplier) to describe every available value while for higher precision resistors you need 3 figures (plus multiplier), see https://en.wikipedia.org/wiki/E-series_of_preferred_numbers#Lists

crazyfermions, 2018-05-17:
I also looked a bit into the code and understand now how it is supposed to work. The first bug is related to the fact that if there is no additional factor like k, M, G in front of Ohm, it returns a null character and the string which is pasted to the clipboard is just terminated.

The other two bugs are related because the colours appear twice and e.g. the two _black_ entries have different meanings; this leads then in some cases to the wrong interpretation for the program. 

@in3otd but the code is still buggy, right? I mean I misunderstood the idea at first because the meanings of the resistor strips change and this is also reflected in the program (an idea which I like in principle, but it requires some additional program logic to guarantee the user does not enter nonsense and this is clearly lacking right now), but clearly the program should not behave as mentioned in bug #3?

in3otd, 2018-05-17:
yep, I was referring to the comment from @andresmmera, about the number of bands being related to the resistor tolerance - the issues you listed above are real, sorry for the confusion.

guitorri, 2018-07-19:
There are nicer resistor calculators available online. How about we drop this feature?

See:

https://www.digikey.be/en/resources/conversion-calculators/conversion-calculator-resistor-color-code-5-band

https://www.allaboutcircuits.com/tools/resistor-color-code-calculator/

andresmmera, 2018-07-20:
> calculator-resistor-color-code-5-band

Honestly, I've never used this tool for the following reasons:

The color code is only used for thru-hole resistors... The vast majority of
resistors out there are SMD type.

The color code table is something simple and easy to learn. In practice,
one only need such tools when you forgot the value of some color...

In this sense, there's no objection from my part to remove the resistor
color code tool.

in3otd, 2018-07-20:
I also never used it, since I learned the color code a loong time ago :grin: and SMDs are more common nowadays.
Keeping it could make more sense if it could also retrieve the resistor data from the schematic, e.g. from  the schematic open the tool showing the color bands corresponding to its value or vice-versa. And maybe add a little more stuff, like parallel/series combinations calculator, resistor dividers calculator, etc.
So at present I have no objection in removing it, maybe for the moment just don't build the tool and keep the sources in the repo, as for the filter v2 tool, in case someone wants to work on the code in the future.

guitorri, 2018-07-20:
Long ago I found this resistor code as a patchset on SF. I merged because I thought it might be useful to beginners. Now I think it is doesn't make sense for us to maintain it, given the alternatives. 

If we don't test it will bit rot. Well, people is still doing global search and replaces in filter-v2 even thou it probably does not even build anymore if you try. I would rather delete. If someday someone has interest, go back in history and figure out how to make things work again. Less is more.

felix-salfelder, 2018-07-20:
On Thu, Jul 19, 2018 at 11:33:07PM -0700, in3otd wrote:
> So at present I have no objection in removing it, maybe for the moment
> just don't build the tool and keep the sources in the repo, as for the
> filter v2 tool, in case someone wants to work on the code in the
> future.

yes. that could be a cool plugin for beginners (once plugins will be
possible..)