Qucs/qucs #803: No digital components and no switch after upgrade

hans005, 2018-06-14:
I ran Qucs in a old version which I do not know what exactly it was (I installed the snap under Xubuntu 17.10) and everything seemed fine until I tried to run a digital project (entering the schematic worked OK, all digital components where present). When trying to simulate that I got an error saying the qucsdigi was not existing.
Since I recently upgrade to Xubuntu 18.04 I thought I'd be clever and upgrade Qucs as well. So I de-installed the snap and re-installed right away.

As a result I now can not open old projects any more since the component "switch" is now missing.
Also I can not load the digital project I created since now ALL the digital components are missing.

Could somebody advice me what has happened here? Did I do something wrong?

Thanks in advance from desperate hans!

in3otd, 2018-06-14:
we do not have official snap packages for Qucs, I guess you maybe installed [this](https://snapcraft.io/qucs-spice). Note that it contains Qucs-S a fork of Qucs from @ra3xdh that allows using `ngspice` and other simulators, besides the native Qucs simulator.

The switch and digital components are built-in in Qucs I cannot see how they could be missing. You may try contacting the snap package creator to see if ey is aware of any issue.

hans005, 2018-06-14:
Ok, I am successfully confused now ;-)
I found Qucs on Sourceforge and also there created an issue because I thought that must then be the fork you mention. But now I guess that is just a Qucs mirror?!

So I also left a message for eldarkg to where your above link points.

Status now is that after again de-installing and re-installing I have the same situation as in the beginning:
I can load my projects again and Switch and digital components are back.
When trying to run a digital simulation I get

ERROR: Cannot start /snap/qucs-spice/4/usr/bin/qucsdigi (No such file or directory)

And indeed that file is missing.

I now downloaded YOUR source files for building it myself but now it is too late and I will  have to try that another time. I will update this ASAP.

Thanks already for your very quick reaction!! :-)

hans005, 2018-06-15:
I tried hard to install qucs (I am NO expert on Linux!):

* got the sources from git
* installed ADMS
* installed dependencies according to the README:
sudo apt-get install build-essential
sudo apt-get install libqt4-dev libqt4-qt3support
sudo apt-get install automake libtool gperf flex bison
sudo apt-get install texlive-font-utils octave-epstk
* installed dependencies according to some forum entry:
sudo apt-get install freehdl
sudo apt-get install libtool-bin
* ./bootstrap
* ./configure --disable-doc (the --disable-doc came from another forum post)
* make
* sudo make install

So finally everything starts up and seems happy except that when I try digital simulation I get

Starting /usr/local/bin/qucsdigi

running C++ conversion...

Errors occurred during simulation on Fr. 15. Jun 2018 at 11:38:40:345
Aborted.

and

Segmentation fault (core dumped)

And I do not know where to find the core dump.....

:-(

felix-salfelder, 2018-06-15:
On Fri, Jun 15, 2018 at 09:56:24AM +0000, hans005 wrote:
> Starting /usr/local/bin/qucsdigi
> 
> running C++ conversion...
> 
> Errors occurred during simulation on Fr. 15. Jun 2018 at 11:38:40:345
> Aborted.
>
> Segmentation fault (core dumped)

Am I right in assuming that only qucsdigi is crashing?

in that case, it would be good to have the crash isolated with an
invocation such as

$ qucsdigi -i mynet.net

(assuming that qucsdigi invocation is similar to qucsator invocation),
you could try and get hold of the "mynet.net" file using the "show
last netlist F6", menu under "Simulation".

if this is possible, please show us that file...

thanks
felix

felix-salfelder, 2025-01-18:
Oh, thanks. This is not the kind of "netlist" we normally want to use. Must have to do with the way freehdl was hacked into qucs (long before my time).

Circuits like the one you have should now work with Gnucsator (possibly subject to missing component models).

Getting VHDL to work with Qucs is more work, and unfortunately I know little about it.