Qucs/qucs #822: Release 0.0.20 checklist

guitorri, 2018-07-19:
- [x] Branch `release-0.0.20`
- [x] Update NEWS.md
- [ ] Update strings/ push to transifex / warn translators / pull update
- [ ] Tag and produce release candidate source package
- [ ] Tag to https://github.com/Qucs/qucs-help
- [ ] Merge `release-` into `master` and `develop`.  
- [ ] Tag `qucs-0.0.20` into `master`.
- [ ] Move the base of all PRs to `develop`
- [ ] Delete `release-0.0.20`
- [ ] Sync master branch on SF repository one last time...
- [ ] Produce source packages, upload to SF
- [ ] Produce Windows and macOS packages, upload to SF
- [ ] Update website, landing and download pages
- [ ] Announce qucs-devel, qucs-help, update SF news
- [ ] Update Wikipedia

andresmmera, 2018-07-21:
@guitorri, @felix-salfelder  there are a few PRs already tested and ready to be merged like  #769, #770, #767 and maybe #448 (I didn't check this latter). What should we do with these PRs? Should we merge them in develop in the short term (next week or so) or just wait until the new Qucs release?

felix-salfelder, 2018-07-21:
#769, #770 look right to me. (the others don't)

i can't test them now. too much on the table.. otherwise i think they could go to develop, either way.

guitorri, 2018-07-24:
@andresmmera I tagged two of the issues to 0.0.21.
Once reviewed, feel free to merge into `develop`.

Rmano, 2018-07-26:
First of all, **1000+ thanks to all the developers of QUCS**. I am using it since, well, forever and it does what it says; at least analog simulations rocks, especially under didactic/education point of view, and the application is way prettier than, say, LTspice. 
 
To the point: shouldn't be the right time to remove that two "0.0." in front of the release? I've used way worst app claiming 4.0... I think that a 1.0 is overdue, or maybe a Mozilla-like versioning (qucs v20, v21 etc..).

andresmmera, 2018-07-26:
I agree with Romano, qucs v20 looks much better than 0.0.20 :-)

guitorri, 2018-07-26:
We should use semantic versioning to correctly convey the development status. https://semver.org/
If we adhere to that the numbers will quickly grow to whatever they need to be.
This requires more planing and communication among developers.

in3otd, 2018-07-26:
> convey the development status

well, as already pointed out elsewhere, the semver page seems more oriented about versioning libraries. I mean, for an application versioning should be more related to user-visible changes, IMHO. E.g. if after refactoring all the code the appearance and functionality remains the same that would just be a patch version increase.
That's to say that we should be well past version 1.0.0 since a long time [already](https://semver.org/#how-do-i-know-when-to-release-100)...

And by the way, we should probably remove `Status: Alpha` from the [SourceForge](https://sourceforge.net/projects/qucs/) page.

> more planning and communication among developers

add also "more developers" to that

Rmano, 2018-12-15:
After more than two years using QUCS in my day-to-day teaching at a University, where it has been greatly useful and has churned happily almost anything I threw at it, I feel I have to formally ask the Developers (yes, capital D) to rethink the release number. 

I know it's true that a rose with any other name will smell the same, but names *do* matter. A release 0.0.20 conveys the idea of a super-alpha software with a lot of bugs and barely usable, which QUCS is not. Yes, it has some rough edge (who hasn't?) and it could be better  (who couldn't), but it's no more a child. It has grown up, it's able to fly mostly well, and I think that the version should convey this. 

Developers, you did a great thing (and are still doing it). Take the plunge and do yourself a gift (it's the season!) and call it 1.0 (or 0.9 if you want --- but I fill that most of the enhancements you are working on sound more like a "wow, multi-sim core, that's 2.0" thing).

Just my 2 cents, but a lot of people feels that QUCS, like the rose,  deserves a better "name" (version).

Rmano, 2018-12-15:
The important thing is, IMHO, removing the double 0 at the start, that normally means that the SW is just alpha/unstable try. What I *feel* right would be to switch to 0.9, and then going 1.0 when the (not so big) list of bug-type issues is fixed. 

There is a very notable precedent... when Linux kernel jumped to 0.9 in 1992. Being there...

Vort, 2019-02-11:
Do test Windows binaries exists for -rc version? I want to check if certain bug from 0.0.19 is fixed.

in3otd, 2019-02-11:
These links point to the latest Qucs binary packages for Windows, automatically built by AppVeyor from the master branch:
- [64-bit version](https://ci.appveyor.com/api/projects/qucs/qucs/artifacts/qucs-win64.zip?branch=master&job=Environment%3A%20MSYSTEM%3DMINGW64%2C%20MBITS%3D64%2C%20MARCH%3Dx86_64)
- [32-bit version](https://ci.appveyor.com/api/projects/qucs/qucs/artifacts/qucs-win32.zip?branch=master&job=Environment%3A%20MSYSTEM%3DMINGW32%2C%20MBITS%3D32%2C%20MARCH%3DI686)

so they currently contain the binaries for the 0.0.20 release candidate.

rjordans, 2019-04-13:
Hey all, thanks for working on this!  I've been using Qucs quite a bit lately and am loving it.

Regarding the 0.0.20 version, do you have an idea yet on the timeline towards the official release?  I've been playing around with the -rc version lately and haven't found issues so far.

guitorri, 2019-04-15:
I am back from vacations. I will make the release in a few days.

Rmano, 2019-04-15:
...think about that release number... 😉
Thanks for the _amazing_ work!

guitorri, 2019-04-16:
Sorry @Rmano , I will not change the release number at this time.

Rmano, 2019-04-16:
@guitorri I supposed it, it was just a bit of wishful thinking... but please consider, after this release, to change it so that QUCS seems the polished product that it is. There are several _commercial_ 1.0+ things that are _really_ worst...

guitorri, 2019-05-19:
Sorry for the delay, I had a lot of things going on these days. 
I will try to rebuild the Windows binary package this week. If things work out I will finalise the release.

guitorri, 2019-05-22:
Qucs 0.0.20-rc2 is tagged and the tarbals are available at:
https://sourceforge.net/projects/qucs/files/qucs/0.0.20/

I noticed I forgot to update the news/changelog... will do for the release.

guitorri, 2019-05-28:
I am struggling a bit to build the Windows binaries. Things changed on the build system and the official Qt4.8.7 is no longer being configured properly. Working on it...

guitorri, 2019-08-16:
I tried again to build the Windows binaries. Not going well. QTDIR needs to be used because pkgconfig is not available with official Qt4.8. QTDIR should still work after pkgconfig was introduced, but it doesn't. To debug the situation I need to bootstrap on the host. I tried with  Windows/MSYS/mingw/qt4 combination as done for the last binary. However MSYS has an old version of automake and it is failing to bootstrap. I tried cross-compiling with Wine but the bootstraping takes too long and so far I didn't manage to make QTDIR work again. The whole point of doing this way is that gcc needs to be shipped to enable free-hdl and the verilog-a features.

Kezii, 2019-08-18:
drop windows support

in3otd, 2019-08-19:
well, of course that would not be fair to our Windows users - which not surprisingly seem to be the vast majority (well above 90 %, judging from the Sourceforge download stats).

@guitorri not sure I understood what the problem is; what's the main difference between the release binary you are trying to do and the AppVeyor build?

ra3xdh, 2019-08-25:
> I tried again to build the Windows binaries. Not going well. 

Is CMake build system operational for Qucs now? You can try to use CMake.  It should work on Wine without MSYS2.

ra3xdh, 2019-08-25:
> well above 90 %, judging from the Sourceforge download stats

Actually the percentage of Windows users is not 90%, because Sourceforge cannot count installations from repositories. The most of Linux users installing Qucs using repositories, not a Sourceforge.

guitorri, 2019-09-01:
@in3otd, the issue is that Freehd and the Verilog-A dynamic loader require a C++ compiler. 
To simplify the packaging and keep binary compatibility I chose to use the official Qt 4.8.7 binaries and its matching Mingw 4.8.2 compiler. 
So, everything is mached to that compiler and I can simply copy the compiler and the freehdl binaries into the package. 
Last time MSYS was used to build the package, see notes on [wiki](https://github.com/Qucs/qucs/wiki/Build-Windows#old-build-instructions---windows-81---mingw-482---qt-486). 
The problem now is that autotools is failing to locate Qt via the QTDIR variable. I cannot fix it on the host system and bootstrap because the available automake in the MSYS env is too old. 
I tried to fix and bootstrap via wine but it was taking too much of my time... 
I tried in the past to use MSYS2 and override the CC CXX (...) variables but it would still not work.

@ra3xdh, Yes CMake should be working. I was thinking about trying with CMake, did not have the time so far. Hopefully this week...

We could simply drop the dependencies and ship the binary from Appveyor and tell people to install MSYS2? If they want those features?
Anyway Freehdl should be deprecated and dropped in favor of Ghdl.
Verilog-a can be left as optional for advanced users with instructions on how to setup the compiler.

Rmano, 2019-09-01:
My opinion is going with @guitorri idea. The loss of Freehdl and Verilog-a is minor compared to the unavailability of a Windows version (I, for example, use QUCS in several classes, and my University's Labs OS is Windows --- can't change it).

It should be spelled out in the program when trying to use it in the Windows version, of course, so that users know what happens. I would like to help, although my C++ is too rusty --- count with me for testing binaries, on the other hand.

in3otd, 2020-02-03:
Is manually updating `automake` in MSYS a possible solution for the configuration issue?

guitorri, 2020-02-03:
I already wasted too much lifetime messing with build systems. If you want to try, be my guest.

Lets try to merge  #953, #951 and #946 and release.

For Windows we take the package from AppVeyor (ASCO already included), manually add qucs-docs, qucs.ba, freehdl (one last time) and link to Mingw 4.8.2 compiler (needed to run freehdl) on the readme and zip the package as in 0.0.19
This way we might miss only only ADMS and dynamic loading of Verilog-A models. Most likely this will not work because qucsator from AppVeyor with g++9 might not load dlls build with g++4.8.

in3otd, 2020-02-08:
I tried cross-compiling via Wine, got to the point where it will try to use `pkg-config` and fail. There are no `.pc` files for that Qt windows version so even if we had `pkg-config` working it will not help much.
I think `QTDIR` does not work anymore because the logic in the `autoconf` is not completely right, it appears to try to use `pkg-config` in any case.

in3otd, 2020-02-11:
FYI, I've ~~wasted~~ spent some time on fixing the win32 build with MinGW 4.8.2. and the Qt 4.8.6 binaries on Wine with your [build script](https://github.com/Qucs/qucs/blob/develop/contrib/windows/qucs-build-win32.sh) and it's working here now. I'll try to clean it up a little and do a PR soon. Not sure how to add freehdl, ADMS, etc. but we can see that later.

guitorri, 2020-02-29:
I just pushed a `release-0.0.20` branch. 
I will try to build the windows binaries and finalize the release.

ra1nb0w, 2021-03-06:
Nice. I just updated macports version to 0.0.20-rc2 and waiting for the final release. Thank you very much for your work!

buenoshun, 2021-07-09:
Hi guys.
This last post is from 2021? The conversation seem to have stopped here about a year ago.
When can we expect the 0.020 windows binary ?
The link posted above with 64 and 32 bit versions is a broken link.
The 0.020 has a new feature that I happen to need, the de-embedding s-parameter model (inverse matrix).

felix-salfelder, 2021-07-09:
Hi buenoshun.

Thanks for your interest. I only have an incomplete answer, but maybe
others can give more detail.

On Fri, Jul 09, 2021 at 06:30:02AM -0700, buenoshun wrote:
> This last post is from 2021? The conversation seem to have stopped here about a year ago.

As far as I know, the focus has shifted from "release 0.0.20" to "move
to Qt>4" and then to implementing "modular Qucs".

> When can we expect the 0.020 windows binary ?

Qucs 0.0.20 has not been released yet. You could try 0.0.20-rc2.

> The 0.020 has a new feature that I happen to need, the de-embedding
> s-parameter model (inverse matrix).

To my knowledge, de-embedding is a Qucsator feature. There is a
standalone Qucsator package now, and a 0.0.20 release. It does not
depend on Qt and may be easier to build.

buenoshun, 2021-07-09:
Hi Felix,
What is qucsator ? I have been using regular QUCS gui on windows for over a decade.
It seems this qucsator package is not a windows binary, neither a GUI. I cannot build/compile software.
The de-embedding would be a component placed in the schematic drawing, similar tot he s-parameter file component.
In this email thread, a guy named Claudio states that the de-embedding was added to qucs:
https://sourceforge.net/p/qucs/mailman/qucs-help/thread/451239769.1028997.1518724483446%40mail.virgilio.it/
Regards,

felix-salfelder, 2021-07-09:
On Fri, Jul 09, 2021 at 10:36:09AM -0700, buenoshun wrote:
> What is qucsator ? I have been using regular QUCS gui on windows for over a decade.

Qucsator is a program that simulates electrical networks. It has been
added to Qucs on 2003-12-21.

> It seems this qucsator package is not a windows binary, neither a GUI.

Correct.

> I cannot build/compile software.

There should be instructions enclosed. Some operating systems and their
users ship or provide or support readily compiled software packages or
build rules for non-developers e.g. [1].

[1] https://aur.archlinux.org/packages/?O=0&K=qucs

ckoegler, 2021-07-16:
Hi Felix,

I would like to help a bit with the qucs roadmap. My c(++) programming is still a bit rusty though. Could you point me to some smaller task, that I could work on?
Best
Carsten

felix-salfelder, 2021-07-16:
On Fri, Jul 16, 2021 at 11:06:21AM -0700, Carsten Kögler wrote:
> I would like to help a bit with the qucs roadmap. My c(++) programming
> is still a bit rusty though. Could you point me to some smaller task,
> that I could work on?

Thanks for your interest. Recently arch users [1] have reported a crash
that needs fixing [2]. I believe that this is worth doing, as any future
qucs will likely run qucsator.

Qucsator has few more bugs of less severity. I don't have a favourite.
Perhaps it would be nice to include the regression tests from qucs-test,
for convenience.

Beyond that, and off topic, I tend to work on modular Qucs these days.
Importing or activating more tests (also from qucs-test) may be a
starting point, and needs doing. Other stuff quickly becomes less small.

[1] https://aur.archlinux.org/packages/qucs/
[2] https://github.com/Qucs/qucsator/issues/29

felix-salfelder, 2024-07-03:
0.0.20 is tagged and archived. Thanks to all who tried.

closed 2024-07-03 by felix-salfelder