Qucs/qucs #828: Compiling 0.0.19 for new Ubuntu: crash with DC simulation

Rmano, 2018-07-22:
Hi, 
I am trying to compile the "master" (stable) branch, which I suppose is 0.0.19, for the new Ubuntu 18.04 Bionic Beaver.  My idea is to try to provide users with something to use while the development is going on; basically, I'd like to create a snap or something like that. I used the PPA for 16.04, but that seemed dead, and I am trying to contribute back something...  

The compilation seems to go well, the program seems to work with my old projects (that worked ok in 0.0.19-rc3 in Ubuntu 16.04, with the FransSchreuder's PPA. The only problem is that the "DC simulation" is not working at all: as soon as there is a DC simulation block in the project, the simulator fails with a segfault. 

Is this problem known? Can any developer help me in solving it? Thanks a lot!

felix-salfelder, 2018-07-22:
On Sun, Jul 22, 2018 at 01:52:41AM -0700, Romano Giannetti wrote:
> Hi,
> I am trying to compile the "master" (stable) branch, which I suppose
> is 0.0.19, for the new Ubuntu 18.04 Bionic Beaver.  My idea is to try
> to provide users with something to use while the [...]

thanks.

the master branch is only used for releases. it might be worth
considering the develop branch for this kind of things. Guilherme is
working on the 0.0.20 release, which will be very close to the current
development.

note that there is preliminary work on a debian package. my idea was
roughly the same: provide something intermediate. we could join forces
in principle. i got a ton of useful (expert) feedback, but i haven't
come to actually to it, as an official package will not be accepted
unless we support qt5...

see https://alioth-lists.debian.net/pipermail/pkg-electronics-devel/2018-June/thread.html

Rmano, 2018-07-22:
My idea was to have something for the users that upgrade from Ubuntu 16.04 to 18.04 --- so that they can continue using QUCS ;-).
I have managed to create a snap package with the Frans Schreuder PPA on my 16.04 machine. I tested it on my 18.04 and it seems to work. You can find details at https://snapcraft.io/qucs-19-rg --- I copied the wrapper script, and I really do not understand it, so any suggestion and help are welcome.

Rmano, 2018-07-22:
Yes, that was what I did firstly. I managed to compile the master branch, and it worked, minus the segmentation fault for the DC simulation.... So I went the "plan B" route with the snap.

I would really prefer to fix the main problem, tough... If someone can help me in debugging it, that is (I am an EE guy, not a CS one...) ;-)

felix-salfelder, 2018-07-22:
On Sun, Jul 22, 2018 at 07:20:05AM -0700, Romano Giannetti wrote:
> I would really prefer to fix the main problem, tough... If someone can
> help me in debugging it, that is (I am an EE guy, not a CS one...) ;-)

you mean, the segfault in qucsator?

try and compile without optimization and with debugging symbols on.

$ make CXXFLAGS=-g\ -O0 install

then use valgrind or gdb to hunt down the crash.

$ valgrind qucsator -i netlist |& less -S
...
(will tell you where invalid memory has been accessed, and (sometimes)
why).

$ gdb --args qucsator -i netlist
> run
...
(lets you inspect the execution stack, and variable values before the
crash)

note that if you run qucsator from the build directory, gdb and valgrind
will not work as expected. that's only a wrapper script..

Rmano, 2018-07-22:
Hmmm... did not build with this flag. 

```
/bin/bash ../libtool  --tag=CXX   --mode=link g++  -g -O0 -Wl,-rpath,/usr/local/lib  -o qucsator ucs.o libqucs.la -lm 
libtool: link: g++ -g -O0 -Wl,-rpath -Wl,/usr/local/lib -o .libs/qucsator ucs.o  ./.libs/libqucs.so -lm
./.libs/libqucs.so: undefined reference to `dlopen'
./.libs/libqucs.so: undefined reference to `dlclose'
./.libs/libqucs.so: undefined reference to `dlerror'
collect2: error: ld returned 1 exit status
Makefile:757: recipe for target 'qucsator' failed
```

Rmano, 2018-07-22:
...and with the normal make it compiles, but at the end `make check` says: 

```
FAIL: tests/basic/u=ri/u=ri@dc.net
PASS: tests/basic/u=ri/u=ri@ac.net
PASS: tests/basic/u=ri/u=ri@tr.net
PASS: tests/basic/u=ri/u=ri@dc+sweep.net
PASS: tests/basic/u=ri/u=ri@ac+sweep.net
PASS: tests/basic/u=ri/u=ri@tr+sweep.net
PASS: tests/basic/u=ri/u=ri@sp.net
PASS: tests/basic/voltagediviser/voltagediviser@tr.net
FAIL: tests/basic/components/capacitor/capacitor@dc.net
PASS: tests/basic/components/capacitor/capacitor@ac.net
PASS: tests/basic/components/capacitor/capacitor@tr.net
PASS: tests/basic/components/spfile/spfile@sp.net
```

felix-salfelder, 2018-07-23:
On Sun, Jul 22, 2018 at 10:33:24AM -0700, Romano Giannetti wrote:
> Hmmm... did not build with this flag.
>
> ```
> /bin/bash ../libtool  --tag=CXX   --mode=link g++  -g -O0 -Wl,-rpath,/usr/local/lib  -o qucsator ucs.o libqucs.la -lm
> libtool: link: g++ -g -O0 [...]

when passing CXXFLAGS?

you found another bug..

Rmano, 2018-07-23:
Yes. It compiles when no flags are passed, it does not if I use `make CXXFLAGS=-g\ -O0`.

guitorri, 2018-07-24:
Did you try with the `release-0.0.20` branch?
Related to the snap package (for qucs-s thou): #765

Rmano, 2018-07-25:
Hi Guillame, 

yes, I used @eldarkg snap cited in #765 as a guidance. I have one now that more-or-less works (`inkscape` refuses to be snapped, so I can save only `png` and `svg`, but well, that's secondary). It can help users to have a version for 18.04 now. It's at https://snapcraft.io/qucs-19-rg.

I tried with release-0.0.20, but `configure` fails with: 
```
checking Qt major version... 4 
./configure: line 17816: syntax error near unexpected token `QTest,'
./configure: line 17816: `    PKG_CHECK_MODULES(QTest, ${qt_name}Test )'
configure: error: ./configure failed for qucs
```
...and I am stuck.

Rmano, 2018-07-25:
After installing `pkg-config` package, and applying this (blind) patch: 

```diff --git a/qucs/configure.ac b/qucs/configure.ac
index e09f583df..969ef2112 100644
--- a/qucs/configure.ac
+++ b/qucs/configure.ac
@@ -236,7 +236,7 @@ case $host_os in
 
     dnl Enable C++11 support
     CXXFLAGS="$CXXFLAGS -std=c++0x"
-    PKG_CHECK_MODULES([QTest], [${qt_name}Test ])
+    PKG_CHECK_MODULES([QT], [${qt_name}Test ])
 
     PKG_CHECK_MODULES([QT], [${qt_name}Core ${qt_name}Gui ${qt_name}Script ${qt_name}Svg ${qt_name}Xml Qt3Support], [
       QT_CFLAGS="$QT_CFLAGS $QT_DEF -DQT_DEPRECATED_WARNINGS -DQT_THREAD_SUPPORT -D_REENTRANT"
```
the configure step succeeded, but then compilation breaks with: 

```
In file included from phaseshifter.h:21:0,
                 from phaseshifter.cpp:18:
component.h:21:10: fatal error: Q3PtrList: No such file or directory
 #include <Q3PtrList>
          ^~~~~~~~~~~
compilation terminated.
```

stuck again ;-)

Rmano, 2018-07-25:
...and that is apparently due to a missing `-I/usr/include/qt4/Qt3Support/` (and I do not know if a corresponding library path) --- but I know nothing of this `configure` stuff, so I really do not know where to touch...

Rmano, 2018-07-25:
Well, hacking away, I found that /by trial and error) that 

```
 make CXXFLAGS=-I/usr/include/qt4/Qt3Support/\ -I/usr/include/qt4/QtGui/\ -I/usr/include/qt4/QtCore/\ -I/usr/include/qt4/QtScript/
```
let the compilation go a bit more, but then it stops with 

```
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -DQT_SHARED -I/usr/include/qt4 -I/usr/include/qt4/QtTest -DQT_NO_DEBUG -DQT_NO_DEBUG_OUTPUT -DQT_DEPRECATED_WARNINGS -DQT_THREAD_SUPPORT -D_REENTRANT -I../../qucs -I/usr/include/qt4/Qt3Support/ -I/usr/include/qt4/QtGui/ -I/usr/include/qt4/QtCore/ -I/usr/include/qt4/QtScript/ -MT spicedialog.lo -MD -MP -MF .deps/spicedialog.Tpo -c spicedialog.cpp  -fPIC -DPIC -o .libs/spicedialog.o
spicedialog.cpp: In constructor ‘SpiceDialog::SpiceDialog(QucsApp*, SpiceFile*, Schematic*)’:
spicedialog.cpp:41:22: error: ‘WDestructiveClose’ is not a member of ‘Qt’
     : QDialog(d, Qt::WDestructiveClose)
                      ^~~~~~~~~~~~~~~~~
```
...re-stuck. Will try to unstuck, but this is more difficult. My C++ is 15-years old...

felix-salfelder, 2018-07-25:
On Wed, Jul 25, 2018 at 12:34:13AM -0700, Romano Giannetti wrote:
> -    PKG_CHECK_MODULES([QTest], [${qt_name}Test ])
> +    PKG_CHECK_MODULES([QT], [${qt_name}Test ])

this patch breaks the QT_* variables, as they will be overwritten.

the purpose of the line was, to set QTest_CFLAGS, QTest_LDFLAGS and such.

hth
felix

Rmano, 2018-07-25:
@felix-salfelder  yep, I supposed it was a wrong way to correct it. And probably the rest of the problems arise from that. What is the correct way to fix it?

Rmano, 2018-07-25:
...and the `WDescrutiveClose` thing arise a lot, so probably it's again introduced by my bad patch. I found a reference to it, btw:    http://doc.qt.io/archives/qt-4.8/porting4.html#

Rmano, 2018-07-25:
It seems that 

```
diff --git a/qucs/configure.ac b/qucs/configure.ac
index e09f583df..09ef805b6 100644
--- a/qucs/configure.ac
+++ b/qucs/configure.ac
@@ -236,7 +236,7 @@ case $host_os in
 
     dnl Enable C++11 support
     CXXFLAGS="$CXXFLAGS -std=c++0x"
-    PKG_CHECK_MODULES([QTest], [${qt_name}Test ])
+    PKG_CHECK_MODULES([QTest], [${qt_name}Test])
 
     PKG_CHECK_MODULES([QT], [${qt_name}Core ${qt_name}Gui ${qt_name}Script ${qt_name}Svg ${qt_name}Xml Qt3Support], [
       QT_CFLAGS="$QT_CFLAGS $QT_DEF -DQT_DEPRECATED_WARNINGS -DQT_THREAD_SUPPORT -D_REENTRANT"
```
fix the configure step (removing the blank).

felix-salfelder, 2018-07-25:
On Wed, Jul 25, 2018 at 01:40:12AM -0700, Romano Giannetti wrote:
> It seems that 
> -    PKG_CHECK_MODULES([QTest], [${qt_name}Test ])
> +    PKG_CHECK_MODULES([QTest], [${qt_name}Test])
> fix the configure step (removing the blank). 

thank you!

Rmano, 2018-07-25:
and....    it works (at least, I can simulate a boosted Howland current pump, which was my test bed...) 

Thanks, @felix-salfelder and @guitorri  for the help.

You should add `pkg-config` to the dependency list in the README file, it seems that newer Ubuntu does not install it by default with build-essentials.

closed 2018-07-25 by Rmano