Qucs/qucs #829: qucs-doc: remove epstk as dependency

guitorri, 2018-07-24:
`epstk` is no longer maintained, hard to get, and perhaps not really needed.

- [ ] Port the figure generation script to plain Octave, without `epstk`:
- [ ] Remove `epstk` checks form build system

```
~/git/qucs/qucs-doc/technical $ git grep eopen .
coplanarzl.m:eopen ("coplanarzl.eps",ePageOrientation,ePageWidth,ePageHeight);
mscomparezl.m:eopen ("mscomparezl.eps",ePageOrientation,ePageWidth,ePageHeight);
mszl.m:eopen ("mszl.eps",ePageOrientation,ePageWidth,ePageHeight);
radius.m:eopen ("radius.eps",ePageOrientation,ePageWidth,ePageHeight);
stabbashford.m:eopen ("stabbashford.eps",ePageOrientation,ePageWidth,ePageHeight);
stabgear.m:eopen ("stabgear.eps",ePageOrientation,ePageWidth,ePageHeight);
stabgearex.m:eopen ("stabgearex.eps",ePageOrientation,ePageWidth,ePageHeight);
stabmoulton.m:eopen ("stabmoulton.eps",ePageOrientation,ePageWidth,ePageHeight);
```

Related:
https://sourceforge.net/p/qucs/discussion/311050/thread/0e653955/
https://github.com/Qucs/qucs/issues/711
https://github.com/Qucs/qucs/issues/495

guitorri, 2020-03-10:
Done for Cmake in 5c4edf58abaef01b2fd49abf11ed237d1ec5ba90 and 2f75aed630bd9d54c0ba181ff52497a92ebb257c

I left Autotools unchanged.

felix-salfelder, 2020-09-27:
... and also texlive-math-extra and pgf are gone (see #991).