Qucs/qucs #830: Replace FreeHDL by GHDL

guitorri, 2018-07-24:
FreeHDL is dead upstream. GHDL is the best alternative.
Replace the simulation engines is as simple as modifying the wrappers scripts.
However the digital.pdf tutorial is written around FreeHDL. What should we about it review to match with GHDL flow?

in3otd, 2018-07-26:
> Replace the simulation engines is as simple as modifying the wrappers scripts.

I wanted to do that in #801 but I know nothing about FreeHDL/GHDL.

> However the digital.pdf tutorial is written around FreeHDL

Maybe we could ask on qucs-help if some user could help us here :grin:

ricardofantin, 2018-08-10:
I can make it. I read the [digital.pdf](http://qucs.sourceforge.net/docs/tutorial/digital.pdf) and the most difficult part is "Limitations" in page 27. In this session the limitations of FreeHDL are listed. And there is a workaround in a example to avoid one limitation, if the FreeHDL is completely removed, maybe a new example will be needed.
There will not be an option to use FreeHDL, only GHDL will be available, right?
Some image should be updated? The new interface change visually something?

guitorri, 2018-08-13:
Nice. Note that I got most of the source schematics used on the digital.pdf from @MikeBrinson. I think that just a few are missing. It would be nice to make sure everything still works.
If that is the case, only GHDL should be supported from this point on.
First step is to make the GHDL wrapper work. Not much visual changes are expected. GHDL must be invoked when a VHDL is set on the Digital Simulation block.

haxscramper-github, 2020-03-26:
I'm not 100% sure that I haven't missed anything, but transition from `freehdl` to `ghdl` (for simulation script) can be done by replacing all code related to freehdl with 

```sh
ghdl -a digi.vhdl
ghdl -e TestBench

TIME_NOSPACE=$(echo $TIME | tr -d ' ' )

./testbench --vcd=digi.vcd --stop-time=$TIME_NOSPACE
```

in `/usr/bin/qucsdigi` (full file is [here](https://gist.github.com/haxscramper-github/43f0e50cf0660b3410a362dc71871a12) (I only commented `freehdl` out, but didn't remove anything))

After that I was able to run examples from [digital.pdf](http://qucs.sourceforge.net/docs/tutorial/digital.pdf) without any issues (when simulation script used `freehdl-v2cc` I was getting segmentation faults on absolutely all simulations).

NOTE: I'm just getting started with digital simulations - I don't want to make full PR (at least right now) because I might've missed something. This solution works for stater examples with truth table - there might be more issues that I don't know of right now.

felix-salfelder, 2020-03-26:
On Thu, Mar 26, 2020 at 08:01:23AM -0700, haxscramper wrote:
> I'm not 100% sure that I haven't missed anything, but transition from
> `freehdl` to `ghdl` (for simulation script) can be done by replacing
> all code related to freehdl with 
> in `/usr/bin/qucsdigi`

perhaps it is adequate to allow alternative "qucsdigi" executables akin
to QUCSATOR. how about QUCSDIGI? then we can have as many *hdls as
anybody would like, and without interference.

also no need to patch qucsdigi after this.

v2e, 2020-04-22:
Not exactly about replacing FreeHDL with GHDL (sorry), but this is related.
@haxscramper-github 
> when simulation script used `freehdl-v2cc` I was getting segmentation faults on absolutely all simulations

Some weeks ago I finally managed to "fix" the FreeHDL 0.0.8.
I gathered and composed a set of patches, and attached them as a [single archive](https://bugs.launchpad.net/ubuntu/+source/freehdl/+bug/1781054/+attachment/5343520/+files/freehdl_0.0.8-2.2ubuntu3.debian.tar.xz) to the [corresponding bugreport](https://bugs.launchpad.net/ubuntu/+source/freehdl/+bug/1781054) on Launchpad. I am not sure my fixes are correct, but at least they let me build and use the FreeHDL-0.0.8 together with QUCS-0.0.20-RC2 successfully.
I also posted a step-by-step description how to build and install the DEB-package for LinuxMint [here](https://forums.linuxmint.com/viewtopic.php?p=1797995&sid=f3ea805afdc6544a074a5ad24a6a8721#p1797995). This should probably also work for any DEB-based distribution.
The problem is that, unfortunately, I did not find any upstream repository for FreeHDL, so I have no Idea where to suggest my patches for other people to have benefit of them.

So having several VHDL options in QUCS is probably a good idea in general, but as of 2020, FreeHDL is still a more-or-less working option.

ddrake, 2021-09-03:
I recently built Qucs 0.0.20 from source, and it's working very well in most cases, but digital simulations failed.  I tried the [FreeHDL 0.0.8 fix](https://forums.linuxmint.com/viewtopic.php?p=1797995&sid=f3ea805afdc6544a074a5ad24a6a8721#p1797995) posted by @v2e, but it didn't build on Ubuntu 20.04 for me (double free detected).  I also tried the GHDL approach outlined by @haxscramper.  I cloned the GHDL repository, built the current master and installed it in /usr/local.  Then from my .qucs directory, I tried these steps suggested above.
```
ghdl -a digi.vhdl
ghdl -e TestBench

TIME_NOSPACE=$(echo $TIME | tr -d ' ' )

./testbench --vcd=digi.vcd --stop-time=$TIME_NOSPACE
``` 
but no testbench script was created so the last command failed.  Please let me know if I've missed something obvious...

felix-salfelder, 2025-01-18:
Preparation of circuits for simulation must (and will) be split from the schematic editor. First of all, the *HDL fragments found in some component drawings (hardwired, c++) need to move to a library. NB: Icarus Verilog: same situation.

This is not really a Qucs issue, but users will expect the Qucs devices in that library. Help is needed to create one, or find one, and set it up.

For simulation there is Gnucap/Gnucsator now anyway (with some of the digital Qucs device models missing, too). It would be nice to have an elementary library for each simulator, including your favourite. With the option to bring-your-own-model that ought to be sufficient.