Qucs/qucs #843: How to write piecewise function in equation-model?

sun0727, 2018-08-07:
Hello everyone,

I would like to ask how to write piecewise funktion like 

x = 1   for  1<= x < 2
x = x   for  2 <= x

Thank you all

in3otd, 2018-08-07:
hello,
if you are referring to the EDD component, take a look at the [Curtice level 1 MESFET model](http://qucs.sourceforge.net/examples/curtice1_MESFET_prj.tar.gz) example on the Qucs website. The `curtice_1.sch` schematic there has an EDD component that implements a piecewise function using the ternary if operator ( *condition* `?` *a* `:` *b* ).

sun0727, 2018-08-07:
@in3otd  thanks for your nice answer

sun0727, 2018-08-07:
@in3otd  

to get function

`res = 19  for  1e6 <=frequency < 10e6`

I wrote

`res = frequency>=1e6 && frequency<10e6 ? 11 : 0`

but always got

```
checker error, no appropriate function for `((frequency>=1e+006)&&(frequency<1e+007))' found
checker error, no appropriate function for `(((frequency>=1e+006)&&(frequency<1e+007))?11:0)' found
checker error, type of equation `res' undefined
```

What's the Problem ?

in3otd, 2018-08-09:
frequency is a vector so the result of the comparisons is a vector too: the two alternatives then need to be vectors too. There is no "broadcasting" of the arguments.
But it's better to explain what you are trying to achieve here, I suspect this may not be the right way.

sun0727, 2018-08-09:
@in3otd  thanks for your anser， I just want to add a reference curve to figure.  The referene curve is a piecewise function, for exampe

res = 19      for        1e6 <=frequency < 10e6
res = 35      for        10e6 =< frequency 

Could you tell me which is the right way for this?

Thanks

in3otd, 2018-08-11:
um, it should be easier but the only way I can think of is something like this
[image]
It's a pity that the *and* (`&&`) operator does not work on vectors.

For more complex reference lines you may also try writing directly the values to a `.dat` *dataset* file and plot it together with your data.