Qucs/qucs #852: DC simulation crashes with some circuit

KevinNadaud, 2018-08-22:
Dear Qucs mainteners,

DC simulation crashes with some circuit, for example this one:

```
R:R1 _net0 _net1 R="15 kOhm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
BJT:T_2N2222_1 _net0 VC _net2 VC Type="npn" Is="1e-14" Nf="1" Nr="1" Ikf="0.3" Ikr="0" Vaf="100" Var="0" Ise="0" Ne="1.5" Isc="0" Nc="2" Bf="200" Br="3" Rbm="0" Irb="0" Rc="3" Re="1" Rb="10" Cje="25e-12" Vje="0.75" Mje="0.33" Cjc="8e-12" Vjc="0.75" Mjc="0.33" Xcjc="1.0" Cjs="0" Vjs="0.75" Mjs="0" Fc="0.5" Tf="400e-12" Xtf="3" Vtf="0.0" Itf="2" Tr="100e-9" Temp="26.85" Kf="0.0" Af="1.0" Ffe="1.0" Kb="0.0" Ab="1.0" Fb="1.0" Ptf="0.0" Xtb="0.0" Xti="3.0" Eg="1.11" Tnom="26.85" Area="1.0"
R:R2 gnd _net0 R="1 kOhm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
R:R3 VC _net1 R="22 kOhm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
R:R4 gnd _net2 R="1.2 kOhm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
C:C1 Ve _net0 C="100 nF" V=""
R:R5 gnd Vs R="1 kOhm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
BJT:T_2N2222_2 VC _net1 Vs _net1 Type="npn" Is="1e-14" Nf="1" Nr="1" Ikf="0.3" Ikr="0" Vaf="100" Var="0" Ise="0" Ne="1.5" Isc="0" Nc="2" Bf="200" Br="3" Rbm="0" Irb="0" Rc="3" Re="1" Rb="10" Cje="25e-12" Vje="0.75" Mje="0.33" Cjc="8e-12" Vjc="0.75" Mjc="0.33" Xcjc="1.0" Cjs="0" Vjs="0.75" Mjs="0" Fc="0.5" Tf="400e-12" Xtf="3" Vtf="0.0" Itf="2" Tr="100e-9" Temp="26.85" Kf="0.0" Af="1.0" Ffe="1.0" Kb="0.0" Ab="1.0" Fb="1.0" Ptf="0.0" Xtb="0.0" Xti="3.0" Eg="1.11" Tnom="26.85" Area="1.0"
Vdc:V2 _net1 gnd U="15 V"
.DC:DC1 Temp="26.85" reltol="0.001" abstol="1 pA" vntol="1 uV" saveOPs="no" MaxIter="150" saveAll="no" convHelper="none" Solver="CroutLU"
Vac:V1 Ve gnd U="10 mV" f="1 k" Phase="0" Theta="0"`
```

When I made the simulation into text mode, I get a segmentation fault. this segmentation fault appears on Linux Mint 19 (based on Ubuntu 18.04) but not on Linux Mint 18 (based on Ubuntu 16.04). 

The thing which is weird is when I add a sweep (even if no parameter is swept), the simulation works well:

`.SW:SW1 Sim="DC1" Type="lin" Param="alpha" Start="0" Stop="1" Points="6"`

I can give you additional information if you need them.

in3otd, 2018-08-22:
which version of Qucs are you using, the `0.0.19` release or the current `develop` branch code?
Your netlist can be simulated fine here locally.
Are you familiar with debugging with `gdb`? If you could `configure` Qucs with `--enable-debug` and compile, then using `gdb` we could get some more information on why the simulator crashes there.

KevinNadaud, 2018-08-22:
I am using the `0.0.19rc1-4` version (the one presented into the ppa of Mint). I am not familiar with `gdb` but I will try to debug it.

in3otd, 2018-08-22:
it could also be related to issue #710. To check the `libc` version use `ldd --version`

KevinNadaud, 2018-08-23:
Here is the output of `$ ldd --version`
`ldd (Ubuntu GLIBC 2.27-3ubuntu1) 2.27`

I think it answers to the question since the version is newer than 2.25. Do you need I try with the development version ?

scordilis, 2018-09-24:
:( I wanted to get back to business using qucs and finishing the qucs workbook.... but .... 
I confirm, I have the issue since I have made the update of my linux box ... .same issue with a .DC and .SP, the engines crashes and with a sweep of the DC voltage (only 2 points) it works again. It really needs to be solved, if not the SP non linear solver is useless.

[buggy.tar.gz](https://github.com/Qucs/qucs/files/2412944/buggy.tar.gz)

`Linux merlin 4.15.0-34-generic #37-Ubuntu SMP Mon Aug 27 15:21:48 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`

in3otd, 2018-09-25:
As mentioned above, please check your system `libc` version, with `ldd --version`: if it is is newer than 2.25 you may see the issue #710.
If you use the current `develop` branch code it should work (well, right now `develop` is broken, as it does not compile but this is another matter, see #871 for the fix)

devel0, 2018-10-20:
glad to hear it works for you,  this mean that there are some dependencies different from mine. I tried DC simulation but that worked. what I'll try in short will to prepare a docker environment that let reproduce consistently any behaviour and as a bonus allow to run within different inside system ( my host is Ubuntu 18.04 ) but within docker I can build and run internally a 17.04 image. I'll let know results.

in3otd, 2018-10-20:
um, very strange. I did not check recently but the first step after creating the netlist for a DC simulation ("`creating node list for DC analysis`") should be almost identical to what's done for a transient simulation.
Can you use `gdb` to check where the loop happens ? (compile `qucs` with the `--enable-debug` option first)

devel0, 2018-10-20:
with `--enable-debug` window show following

[image]

while attaching vscode to `ucsator` process gives endpoint after max iterations

[image]

reached by followings principals:

[image]

in particular, apart it doesn't solve it, seems can't exit [this loop](https://github.com/Qucs/qucs/blob/5896b2480fc3eef7ad76780f053df7f7f01b24d4/qucs-core/src/trsolver.cpp#L246) due to saveCurrent not proceed toward time limit

[image]

As already said I'll try now compiling either in Ubuntu 17.04 and 18.04 through a docker container and I'll let know if it works

in3otd, 2018-10-20:
the transient simulation is not very robust and can get stuck like you saw. Unfortunately a general solution is not straightforward.

But the initial issue was described as "got infinite loop on `creating netlist`": was then the behavior different without enabling debug?

devel0, 2018-10-20:
no, it's the same, or in other words with or without `--enable-debug` it not proceed over "creating netlist" but with debug at least some useful warnings appears. The strange thing was that some day ago I was able to run transient solutor on the schematic given before I made some changes that I don't remember ( if I find which one was I'll let know about ). thanks you for directions about debug switch.

felix-salfelder, 2018-10-20:
On Fri, Oct 19, 2018 at 11:38:43PM -0700, in3otd wrote:
> um, very strange. I did not check recently but the first step after
> creating the netlist for a DC simulation ("`creating node list for DC
> analysis`") should be almost identical to what's done for a transient
> simulation.

there's a switch involved with switchtime set to 0ms
does the model evaluate differently in dc vs tran@0s?

some spice models behave weird during events, maybe the switch does that
too?

another idea: if a dc analysis does not converge, the "result" might
look plausible. usually, attempting a transient from such a solution
fails: the time derivatives don't work.

devel0, 2018-10-20:
I already think of that I tried removing switches results doesn't change,
I also tried a [ubuntu 17.04](https://github.com/SearchAThing-forks/qucs/tree/docker/docker/xenial) version with same result regardless ldd version was 2.23

felix-salfelder, 2018-10-21:
On Sat, Oct 20, 2018 at 09:59:52AM -0700, in3otd wrote:
> the transient simulation is not very robust and can get stuck like you
> saw. Unfortunately a general solution is not straightforward.

I know it's cheating. your simulation works in gnucsator with the bjt
model from ngspice17. put a "load models/ngspice17/bjt.so" into your
gnucsator.rc...

(background: the gnucap builtin modelgen (pre-verilog-A) bjt model has
been superseded and discontinued when this mechanism was ready.)

the hand written models for spice are full of hacks and convergence
related tricks, which you don't want to mess with in either verilog-A or
modelgen. sometimes this helps, sometimes this produces wrong
results. ymmv.

> But the initial issue was described as "got infinite loop on `creating
> netlist`": was then the behavior different without enabling debug?

its a very plausible convergence issue. this kind of stuff happens if
you leave your collector floating...

devel0, 2018-10-21:
I'm sorry I realized now that this was not the correct issue for reporting problem about transient BJT that is instead issue #41 ; btw I confirm that using different transistor model 2N3906 in place of a BC557B it works ; working sim circuit can be found [here](https://github.com/devel0/iot-cook-timer/tree/f171f94944015eefacb108a7d7aab961d8e653bd#power-management-circuit). More I'm about a newbie in electronics but real circuit measures a voltage of 3.02V at Vm instead of 2.64V. [sch](https://github.com/devel0/iot-cook-timer/blob/f171f94944015eefacb108a7d7aab961d8e653bd/doc/prj.sch)

felix-salfelder, 2018-10-21:
thank you for referencing #41. this looks related.

FWIW: i get 0->3.58V Vav and 3.75->0V Vm  (transition at 1e-4s), using BC557B.