Qucs/qucs #860: install with cmake is broken

in3otd, 2018-09-04:
when running `make install` after running `cmake`
- it fails because it does not find the file `qucs-powercombining/qucspowercombining.1`
@andresmmera did you forget to upload the `man` page for `qucs-powercombining`? :grin:
Or should we remove that reference (the autotools build doesn't try to install that `man` page)
- after fixing the point above, `qucs` does not run because it does not find the `libqucsschematic.so` installed in `<install_prefix>/lib`

On Travis we do a build using `cmake` also but we don't do a `make install` there so these errors do not occur.

andresmmera, 2018-09-04:
> it fails because it does not find the file qucs-powercombining/qucspowercombining.1
@andresmmera did you forget to upload the man page for qucs-powercombining? grin
Or should we remove that reference (the autotools build doesn't try to install that man page)

Honestly, I didn't write any man page... but I can do it soon. Gimme a few days :-)

andresmmera, 2018-09-05:
Ok, I was taking a look to the man page files of the other tools and I think this text is outdated:

> The latest version of Qucs can always be obtained from
www.sourceforge.net\ or www.freshmeat.net

I've browsed www.freshmeat.net and it seems there's nothing related to Qucs there... Shall we remove that and mention that the source code is available on Github?

felix-salfelder, 2018-09-05:
On Wed, Sep 05, 2018 at 11:05:01AM -0700, Andrés Martínez-Mera wrote:
> Ok, I was taking a look to the man page files of the other tools and I think this text is outdated:
>
> www.sourceforge.net\ or www.freshmeat.net
>
> I've browsed www.freshmeat.net and it seems there's [...]

i'd prefer no link over an outdated one. the more links we have, the
less likely they will be maintained.

(how about @QUCS_URL@ everywhere and a top level URL file,
similar to VERSION, see [1]. but i am a bit biased.)

[1] https://en.wikipedia.org/wiki/Don%27t_repeat_yourself

andresmmera, 2018-09-08:
> i'd prefer no link over an outdated one. the more links we have, the less likely they will be maintained.

I see your point, but if we include a section called 'Availability' in the manpage, we should provide some links to the readers.

> (how about @QUCS_URL@ everywhere and a top level URL file, similar to VERSION, see [1]. but i am a bit biased.)

I fully agree. In this sense, I created a new file at /qucs/man/ to avoid repeating the same text in every
manpage file: qucs_availability_bugs.1

Concerning the 'Availability' stuff, I replaced "www.sourceforge.net" by "http://qucs.sourceforge.net/" and "www.freshmeat.net" by "https://github.com/Qucs/qucs\"

I noticed that we repeat the same description of the Qucs software in every manpage. Again, following the DRY principle, I tried to put that stuff in a common file. Surprisingly, it seems not possible call the ".so" command twice in the same file. Does anyone have expertise with this?

See PR #862

felix-salfelder, 2018-09-08:
On Sat, Sep 08, 2018 at 12:58:13AM -0700, Andrés Martínez-Mera wrote:
> [..] I created a new file at /qucs/man/ to avoid repeating the same text in every
> manpage file: qucs_availability_bugs.1

sounds like a good approach.

> I noticed that we repeat the same description of the Qucs software in
> every manpage. Again, following the DRY principle, I tried to put that
> stuff in a common file. Surprisingly, it seems not possible call the
> ".so" command twice in the same file. Does anyone have expertise with
> this?

I did not know about .so, nor is it documented in my man(7).

FWIW: my approach to this would be more simplistic. just move file to
file.in, and add it to AC_CONFIG_FILES. you get all the @VARIABLES@
substituted. of course that does not play well with whole paragraphs.
YMMV.

guitorri, 2018-09-11:
@andresmmera are you also looking into the missing <prefix>/lib issue?

guitorri, 2018-09-14:
Cmake Installation should be fixed. Awaiting your manpages and further autotools fixups from #862. I will take care of propagating the variable substitutions with cmake.

closed 2018-09-14 by guitorri

andresmmera, 2018-09-15:
>  I will take care of propagating the variable substitutions with cmake.

Oh thanks!! I did a first attempt (with no success). It went fine for the top-level directory, but I didn't find the way to propagate the variables to the other CMake files.  https://github.com/andresmmera/qucs/commit/2439bed8477f60dba793ecc0d991587513de5242
The get_directory_property() macro didn't work for me... just scratching my head about how to do that...

andresmmera, 2018-09-15:
Ok, I got it. I'll push a commit to #862 Feel free to remove that if you consider it's not correct