Qucs/qucs #861: Optimization is still broken

in3otd, 2018-09-07:
the fix in #800 is not enough; the optimization runs but the final results are not updated in Qucs due to a missing last step in the "general optimization" mode of ASCO (w.r.t. the "Qucs mode")

guitorri, 2018-09-27:
Any progress on this one?

in3otd, 2018-09-27:
yes, I've tested #865 on Linux and Windows and it works fine here. Still I'd like to do more testing to avoid having to fix it a third time :grimacing:

in3otd, 2019-01-08:
fixed by #865

closed 2019-01-08 by in3otd

in3otd, 2019-01-15:
Does the optimization appear to progress normally? In the Simulation Messages window you should see values for *Best-so-far cost funct. value* and *cost-variance* which are not too big. If they stay at e.g. 1e31 something is wrong with the optimization and in these cases I've seen that when it stops due to the maximum number of iterations reached there are no results available.

Also, just to be sure, did you patch ASCO as per #865? Anyway, if you have `qucs` in the path this may not be needed. "fresh build" means you are on which commit? Can you share the failing example?

michal777, 2019-01-16:
I compared outputs (F5 key) from qucs and qucs-s - they are identical but the correct results are only available in qucs-s
Now I've got ASCO 0.4.10 downloaded from the http://asco.sourceforge.net/ and qucs cloned today from this repo (Latest commit d25203a 3 days ago). I assumed the #865 is merged in this repo so I needn't care.
Do I miss something?
The example is simple voltage divider.
[tmp.sch.txt](https://github.com/Qucs/qucs/files/2766267/tmp.sch.txt)
[qucs_optim_log.txt](https://github.com/Qucs/qucs/files/2766268/qucs_optim_log.txt)

in3otd, 2019-01-17:
ok, I can reproduce it also here, if I do not patch ASCO. Due to the way now the ASCO optimizer is called you *do* need the [patch](https://github.com/Qucs/qucs/blob/develop/contrib/ASCO-Qucs_simulator_name.patch) to ASCO 0.4.10 which was added in #865.

guitorri, 2019-01-18:
@in3otd did you send the patch upstream?

in3otd, 2019-01-18:
no, I didn't. But the patched ASCO should still work with the pre-#730 Qucs (and likely with Qucs-S) - I did not actually check yet - so it may make sense to have it included upstream.

dom11990, 2020-08-05:
I'm not able to get this to work... maybe i am doing something wrong but when I try to apply the patch?

```
patch --dry-run -p1 < ASCO-Qucs_simulator_name.patch 
checking file asco-test.c
Hunk #1 FAILED at 29.
Hunk #2 FAILED at 179.
2 out of 2 hunks FAILED
checking file asco.c
checking file errfunc.c
```