Qucs/qucs #880: Convergence problem with no GND

algorithm314, 2018-11-17:
I Qucs there is convergence problem in transient simulation when ground is not present especially for non linear components. As most new users(like me) do not specify ground as it is not mandated, I propose Qucs should display error if GND is not present.

felix-salfelder, 2018-11-17:
On Sat, Nov 17, 2018 at 04:48:03AM -0800, algorithm314 wrote:
> I Qucs there is convergence problem in transient simulation when
> ground is not present especially for non linear components. As most
> new users(like me) do not specify ground as it is not mandated, I
> propose Qucs should display error if GND is not present.

the presence of GND it neither required nor sufficient for convergence.
it is clearly better not to display an error. the error is the
non-convergence.

perhaps qucsator should print "notice: no components are connected to
ground, this may affect convergence in some situations.", in case no
components are connected to ground.

hth
felix

algorithm314, 2018-11-17:
Yes something should be displayed. But it is very strange that even a single diode rectifier conected to a resistor has problems with convergence. Also in my opinion the installation process of qucs is very difficult for Linux. I could only do it through source. A self contained directory will be very helpfull.

algorithm314, 2018-11-17:
The circuit is this 
[image] and it works.But without the GND it does not work.
The netlist is(including the GND)
```
# Qucs 0.0.19  /home/konstantinos/.qucs/test.sch

.TR:TR1 Type="lin" Start="0" Stop="100 ms" Points="100" IntegrationMethod="Euler" Order="2" InitialStep="1 ns" MinStep="1e-5" MaxIter="150" reltol="0.001" abstol="1 pA" vntol="1 uV" Temp="26.85" LTEreltol="1e-3" LTEabstol="1e-6" LTEfactor="1" Solver="CroutLU" relaxTSR="no" initialDC="yes" MaxStep="0"
Diode:D_1N4148_1 _net0 _net1 Is="222p" N="1.65" Cj0="4p" M="0.333" Vj="0.7" Fc="0.5" Cp="0" Isr="0" Nr="2" Rs="68.6m" Tt="5.76n" Ikf="0" Kf="0" Af="1" Ffe="1" Bv="75" Ibv="1u" Temp="26.85" Xti="3.0" Eg="1.11" Tbv="0.0" Trs="0.0" Ttt1="0.0" Ttt2="0.0" Tm1="0.0" Tm2="0.0" Tnom="26.85" Area="1.0"
R:R1 gnd _net0 R="50 Ohm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
VProbe:Pr1 _net0 gnd
Vac:V1 _net1 gnd U="50 V" f="50 Hz" Phase="0" Theta="0"
.DC:DC1 Temp="26.85" reltol="0.001" abstol="1 pA" vntol="1 uV" saveOPs="no" MaxIter="150" saveAll="no" convHelper="none" Solver="CroutLU"
```
With self contained directory i mean a directory that contains all required dependencies. Like windows installation or the same as when you download firefox for Linux from the site.

As for the dificulties the packages names are not the same(for example libgd-dev instead of libgd2-xpm-dev) and adms libraries when installed where not in the PATH. Also i dont think many new users will do all the required steps.