Qucs/qucs #882: PDF bounding box wrong

Rmano, 2018-11-29:
Hi, 
I have (I do not know how) managed to create a `.sch` file which has the bounding box of the exported PDF wrong. Exporting in `png` and `eps` works ok, but the PDF is quite strange: 

[image]

I am using: `Qucs 0.0.20 (a7597048f)` (compiled here, Ubuntu 18.04). 

Files: /home/romano/lib/qucs/sallenkey_prj/BW3k-ok.sch
[test.zip](https://github.com/Qucs/qucs/files/2629198/test.zip)

in3otd, 2019-01-10:
yes, can be reproduced also here: with *File->Export as image...* you get a `pdf` with the wrong bounding box. It works fine when using *File->Print...*. Seems to have something to do with using `inkscape` for the conversion, I'm taking a look.

closed 2019-01-13 by guitorri

Rmano, 2019-01-15:
Sorry for the delay; checked, works ok. Thanks.!

PR #896 (2019-01-13): Fix PDF bounding box
Fixes #882. 

When creating a PDF file using *File->Export as image...* the resulting PDF size was often not correct.
Now the bouding box is specified in the intermediate `svg` file and not left to `inkscape` to guess/compute.

I've tested it on a few examples and seems to work fine but please double check.