Qucs/qucs #902: power probe problems?

Rmano, 2019-01-29:
So, I probably am in for a scolding, but... can anyone explain to me this? Shouldn't Pr1.P equal to Pr1.V multiplied Pr1.I? 

[image]

The netlist is: 
```
# Qucs 0.0.20  /home/romano/lib/qucs/InEleSAP_prj/test_pprobe.sch

Idc:I1 _net0 gnd I="0.222"
WProbe:Pr1 _net0 _net1 _net0 _net2
.DC:DC1 Temp="26.85" reltol="0.001" abstol="1 pA" vntol="1 uV" saveOPs="no" MaxIter="150" saveAll="no" convHelper="none" Solver="CroutLU"
R:R1 _net1 _net3 R="6" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
IProbe:Pr2 _net3 _net2
VProbe:Pr3 _net1 _net2
R:R2 gnd _net2 R="50 Ohm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
```

felix-salfelder, 2019-01-29:
On Tue, Jan 29, 2019 at 02:25:13AM -0800, Romano Giannetti wrote:
> So, I probably am in for a scolding, but... can anyone explain to me
> this? Shouldn't Pr1.P equal to Pr1.V multiplied Pr1.I? 

in wprobe.cc

line 57, the voltage is computed, stored as Vr. in line 68 it should be
used... possibly something else is wrong.

Rmano, 2019-01-29:
hmmm: 
```
void wprobe::saveOperatingPoints (void) {
  nr_double_t Vr = real (getV (NODE_3) - getV (NODE_4));
  nr_double_t Vi = imag (getV (NODE_3) - getV (NODE_4));
  setOperatingPoint ("Vr", Vr);
  setOperatingPoint ("Vi", Vi); //This section works just like a voltmeter
}

//For specific information regarding The Power triangle and Power factor:
//https://en.wikipedia.org/wiki/Power_factor#Definition_and_calculation

void wprobe::calcOperatingPoints (void) {
//Reading the current and voltage values to calculate power values
  nr_double_t VAr = real (getV (NODE_3) * getJ (NODE_1));
  nr_double_t VAi = -imag (getV (NODE_3) * getJ (NODE_1));
```
Could be a node name typo? Where are the NODE_x correspondences defined?

felix-salfelder, 2019-01-29:
On Tue, Jan 29, 2019 at 02:54:27AM -0800, Romano Giannetti wrote:
> hmmm:
> ```
> void wprobe::saveOperatingPoints (void) {
>   nr_double_t Vr = real (getV (NODE_3) - getV (NODE_4));
>   nr_double_t Vi = imag (getV (NODE_3) - getV (NODE_4));
>   setOperatingPoint ("Vr", [...]

don't know.

just do it as in line 57, it seems correct. perhaps you need to multiply
first, and then take the real part. so you can't just use Vr.

Rmano, 2019-01-29:
[image]

The correct formula should be: 
```
VAr = real ( (getV(NODE_C)-getV(NODE_D)) * (current through AB) )
```
and the imaginary part the same.

Rmano, 2019-01-29:
This patch seems to fix it, and it feels right: 
```
@@ -65,8 +65,8 @@ void wprobe::saveOperatingPoints (void) {
 
 void wprobe::calcOperatingPoints (void) {
 //Reading the current and voltage values to calculate power values
-  nr_double_t VAr = real (getV (NODE_3) * getJ (NODE_1));
-  nr_double_t VAi = -imag (getV (NODE_3) * getJ (NODE_1));
+  nr_double_t VAr = real ((getV (NODE_3) - getV(NODE_4)) * getJ (NODE_1));
+  nr_double_t VAi = -imag ((getV (NODE_3) - getV(NODE_4)) * getJ (NODE_1));
   setOperatingPoint ("VAr", VAr);
   setOperatingPoint ("VAi", VAi);
```

[image]

Rmano, 2019-01-29:
Would you like to have a pull request? Against which branch? I think this is quite urgent...

closed 2019-02-02 by guitorri