Qucs/qucs #906: Adding a tutorial to documentation

Rmano, 2019-02-02:
Hi, 
I have created a small tutorial for my students on how to use QUCS for analyzing frequency response of linear circuits which includes a step-by-step guide on subcircuits. 
I would like to contribute that to the set of documents --- I was looking through them but it seems not so straightforward... is there a kind of guide/help on how to do it? 
Thanks!

Rmano, 2019-02-02:
I tried to clone the directory `...qucs-docs/tutorial/dcstatic`, changing the name, substituting the relevant information in the file `Makefile.am` and `pics/Makefile.am`, and adding the new directory to `...tutorilas/Makefile.am` --- but I am stuck because I am not able to trigger the build of `Makefile.in` and `Makefile`.

felix-salfelder, 2019-02-02:
On Sat, Feb 02, 2019 at 02:32:07PM -0800, Romano Giannetti wrote:
> I tried to clone the directory `...qucs-docs/tutorial/dcstatic`, changing the name, substituting the relevant information in the file `Makefile.am` and `pics/Makefile.am`, and adding the new [...]

look at configure.ac. it has a list of makefiles to generate. run ./bootstap after changing anything there.

Rmano, 2019-02-03:
Thanks --- done!

closed 2019-02-03 by Rmano