Qucs/qucs #908: adding files to gitignore?

Rmano, 2019-02-03:
After a clean build, `git status` says: 

```
Untracked files:
  (use "git add <file>..." to include in what will be committed)

	qucs-core/doc/qucsator.1
	qucs-core/doc/qucsconv.1
	qucs-doc/report/activefilter/activefilter.pgf-plot.gnuplot
	qucs-doc/report/activefilter/activefilter.pgf-plot.table
	qucs/man/qucs.1
	qucs/man/qucsdigi.1
	qucs/man/qucsdigilib.1
	qucs/man/qucsveri.1
	qucs/qt_version
	qucs/qucs-activefilter/qucsactivefilter.1
	qucs/qucs-attenuator/qucsattenuator.1
	qucs/qucs-filter/qucsfilter.1
	qucs/qucs-lib/qucslib.1
	qucs/qucs-powercombining/qucspowercombining
	qucs/qucs-powercombining/qucspowercombining.1
	qucs/qucs-rescodes/qucsrescodes.1
	qucs/qucs-transcalc/qucstrans.1
	qucs/qucs/qucs.real
	qucs/stamp-h2
	qucs/test-driver
```

Should they be added to the `.gitignore` file(s)? I can prepare a PR if it's deemed interesting.

Thanks!

felix-salfelder, 2019-02-03:
On Sun, Feb 03, 2019 at 11:40:39AM -0800, Romano Giannetti wrote:
> After a clean build, `git status` says: 
> 
> ```
> Untracked files:
>   (use "git add <file>..." to include in what will be committed)
> 
> 	qucs-core/doc/qucsator.1
> [..]
> ```
> 
> Should they be added to the `.gitignore` file(s)? I can prepare a PR if it's deemed interesting.

perhaps the changes should go into qucs/.gitignore, qucs-core/.gitignore
etc. these directories will (likely) be turned into independent
submodules or renamed at some point.

Rmano, 2019-02-04:
> perhaps the changes should go into qucs/.gitignore, qucs-core/.gitignore etc. these directories will (likely) be turned into independent submodules or renamed at some point.

Ok, I'll try. I have doubt about the `qucs/test-driver` directory, which if I've got it, should host a submodule... should I put it in the `.gitignore` or not?

Rmano, 2019-02-04:
...and also the (empty) directory that is created in `qucs/qucs-powercombining/`:
```
-rwxr-xr-x 1 romano romano 1877368 Feb  2 23:34 qucs/qucs-powercombining/qucspowercombining
```
?

closed 2020-02-15 by guitorri