Qucs/qucs #911: Upside down text for mirrored transformer

Vort, 2019-02-11:
1. Place Symmetric Transformer.
2. Mirror it with Ctrl+M.
3. Save schematic.
4. Close schematic.
5. Open schematic again.

T1 and T2 labels becomes upside down.

[image]

Tested with QUCS 0.0.20 (4205dd365) on Windows 7 SP1 x64.

Rmano, 2019-02-11:
Happens the same in Linux; confirmed to be not Windows-specific.

in3otd, 2019-02-13:
I've checked the past Qucs releases and this issue has been there since quite some time, the oldest I tried was 0.0.15 and it behaves the same.

felix-salfelder, 2025-01-18:
Now that we have moved past Qt3/4 it might make sense to refactor symbols and de-duplicate code. It seems that several symbols implement coordinates, drawing, geometry, text in different ways. :(

MultiViewComponent must be renamed to reflect its purpose (if any), or removed.