Qucs/qucs #912: Console window is visible on Windows

Vort, 2019-02-11:
It is common that GUI programs have no console visible in Windows.
It was so in 0.0.19, but in latest build of 0.0.20 console is appearing.

Tested with build ([4205dd3](https://github.com/Qucs/qucs/commit/4205dd3650fe469151fa6932664797b1ed044a9c)) on Windows 7 SP1 x64.

[image]

Vort, 2020-04-04:
> which is the difference between a "console application" and a "graphic application"

The difference is in subsystem flag inside PE (portable executable) file.
You can even change it manually with hex editor inside exe and console window will disappear.
Don't know why cmake documentation focuses on entry point differences.

And that's why another solution is specifying subsystem manually:
`set(CMAKE_EXE_LINKER_FLAGS_RELEASE "${CMAKE_EXE_LINKER_FLAGS} /SUBSYSTEM:WINDOWS /ENTRY:mainCRTStartup")`

> Did you build using CMake or Autotools?

I did not built it at all. Just downloaded binary somewhere.

thomaslepoix, 2020-04-04:
Interesting.

Microsoft [`/ENTRY`](https://docs.microsoft.com/en-us/cpp/build/reference/entry-entry-point-symbol?view=vs-2019) and [`/SUBSYSTEM`](https://docs.microsoft.com/en-us/cpp/build/reference/subsystem-specify-subsystem?view=vs-2019) docs say that `/SUBSYSTEM:WINDOWS` comes by default when `WinMain` is set as entry point.

The MinGW equivalent flag seems to be `-mwindows`.

CMake build [seems good to me](https://github.com/Qucs/qucs/search?q=add_executable&unscoped_q=add_executable), except maybe [__qucs-filter-v2__ tool](https://github.com/Qucs/qucs/blob/d3218d9dadacaa8752f1dc1387be7e46b6ab474d/qucs/qucs-filter-v2/CMakeLists.txt#L57) if launched standalone. Could someone confirm?

By reading Qucs [build instructions](https://github.com/Qucs/qucs/wiki/Build-Windows) and [appveyor file](https://github.com/Qucs/qucs/blob/develop/.appveyor.yml) I assume your binary was built using Autotools but this has to be confirmed too.

I don't believe that Autotools provide a compiler independent flag for that, like CMake do, but I don't know Autotools.

As non Windows user, I cannot help much more..

Vort, 2020-04-05:
> By reading Qucs build instructions and appveyor file I assume your binary was built using Autotools but this has to be confirmed too.

I have found which link I was using.
Here it is: [x64](https://ci.appveyor.com/api/projects/qucs/qucs/artifacts/qucs-win64.zip?branch=master&job=Environment%3A%20MSYSTEM%3DMINGW64%2C%20MBITS%3D64%2C%20MARCH%3Dx86_64).

> I don't believe that Autotools provide a compiler independent flag for that

But it for sure contains conditional directives.

---

However, now that binary is based on different commit, 10ba944c8, which is _completely_ broken.
[And no one cares.](https://sourceforge.net/p/qucs/discussion/311050/thread/456c3abf4e/)

thomaslepoix, 2020-04-06:
An appveyor artifact, so it is Autotools based.

> But it for sure contains conditional directives.

Of course!

> However, now that binary is based on different commit, 10ba944, which is completely broken.

As our problem comes from build systems, this does not affect its resolution, as long as it is possible to build Qucs.

May I ask you to build head version using [Autotools](https://github.com/Qucs/qucs/wiki/Build-Windows#build-instructions) and [CMake](https://github.com/Qucs/qucs/wiki/Build-Windows#build-with-cmake--cmd-terminal) to check if the problem persist and try a fix if it does for the Autotools one?

Vort, 2020-04-06:
> May I ask you to build head version using Autotools and CMake to check if the problem persist

Build instructions are outdated. Both cmake and autotools.
Also project uses outdated Qt version (which support is dropped in MSYS2, but luckily it is still possible to install it manually: [mingw-w64-x86_64-qt4-4.8.7-4-any.pkg.tar.xz](http://repo.msys2.org/mingw/x86_64/mingw-w64-x86_64-qt4-4.8.7-4-any.pkg.tar.xz)).
That is, of course not the only problem. Next one:
[image]
It is fixed trivially, by replacing `LoadLibrary` with `LoadLibraryW` (do anyone want to make PR to fix this problem?).
Next problem is resurrected (who knows how) #295. I was forced to install `dos2unix` and apply fix manually.
Also I was not able to make `-G "MinGW Makefiles"` work, so replaced it with `-G "MSYS Makefiles"`.
Autotools portion is having problems too: `autogen.sh` is missing. `./bootstrap` should be used instead.
Not mentioning lots of missing dependencies, which was required for build, but was not listed in documentation.

But anyway, 955ccd1b1 is built (at least, partially) and assumption is confirmed: autotools version is having console problem, cmake version is not having it:
[image]
What change do you think I should make to autotools build system to check if it will remove console window?

thomaslepoix, 2020-04-06:
Such an adventure.. well done!

I think it should check which compiler is used :
- MSVC : add `/ENTRY:WinMain` flag, or maybe `/SUBSYSTEM:WINDOWS`
- MinGW (whether the unix compatibility layer is MSYS, cygwin, or something else) : add `-mwindows` flag
- gcc or clang : we are on unix platform, nothing to add

I don't really know if they are compiler or linker flags. I think Qucs GUI and the different tools (transcalc, active-filter, etc.) need the fix.

May you starting point be near [this kind of lines](https://github.com/Qucs/qucs/blob/develop/qucs/configure.ac#L216)?

Note : Actually I am not sure that MSVC is supported, if you cannot build with it, no need to treat its case.

Vort, 2020-04-07:
> I think it should check which compiler is used :

And this part is broken too!

> I don't really know if they are compiler or linker flags.

Linker.

---
Here is what I have done:
```diff
diff --git a/qucs/configure.ac b/qucs/configure.ac
index 3d32c8a60..148ab5bd4 100644
--- a/qucs/configure.ac
+++ b/qucs/configure.ac
@@ -217,10 +217,10 @@ fi # if pkg-config exists
     *linux* | *cygwin* | *mingw* | *msys* | *bsd* )
 
       dnl Set Clang
-      if test "$CXX -dM -E - < /dev/null | grep __clang__" ; then
+      if `$CXX -dM -E - < /dev/null | grep __clang__` ; then
         CFLAGS="$CFLAGS -pipe"
         CXXFLAGS="$CXXFLAGS -pipe -fno-exceptions -Wno-deprecated-register"
-        LDDFLAGS="$LDDFLAGS"
+        LDFLAGS="$LDFLAGS"
         use_CLANG="yes"
       fi
 
@@ -230,6 +230,7 @@ fi # if pkg-config exists
         CXXFLAGS="$CXXFLAGS -pipe -fno-exceptions -fno-check-new"
         if test x$WIN32 = xyes; then
           CXXFLAGS="$CXXFLAGS"
+          LDFLAGS="$LDFLAGS -mwindows"
         fi
       fi
```
And it works (no console window). Except that I did not tested it with clang.

thomaslepoix, 2020-04-08:
Really nice!

Clang build is successful for me, but your fix is partial, use this line instead and it will be ok.

```sh
      if test "`$CXX -dM -E - < /dev/null | grep __clang__`" ; then
```

Vort, 2020-04-08:
https://ci.appveyor.com/api/buildjobs/4hquup1h4149wmwp/artifacts/qucs-win64.zip looks good.
Thanks!

Vort, 2020-04-08:
By the way, Clang build do not produces schematic UI corruption:
[image]

in3otd, 2020-04-08:
Thanks! I think we had the `-mwindows` flag in the past but likely it got accidentally removed when reworking `configure.ac`..

Great that the `clang` build does not have graphic issues: we knew that the AppVeyor build is badly broken (#958) but nobody did find a solution for that yet and to get a working Windows executable currently we had to use Wine (#962).

Maybe you could update the AppVeyor build to use `clang` (in another PR) ? :grin:

closed 2020-07-28 by felix-salfelder

ra3xdh, 2022-09-14:
@Andronx you may try Qucs-S instead https://github.com/ra3xdh/qucs_s/releases/tag/0.0.24 Qucs-S uses Ngspice as the default simulation kernel, but the usage of qucsator is also possible if necessary.

Andronx, 2022-09-15:
@ra3xdh Yes, I know Qucs-S, but Optimization with preffered numbers E24.. so on, functionality is very usefull for me and I didn't found that in Qucs-S or Qucs Studio. That's the reason that I'm looking for Qucs 0.0.20 rc2.

commit 84ae6e41 (2020-07-28): Set correct subsystem value for Windows Autotools build

Fixes #912.
PR #999 (2020-07-28): Set correct subsystem value for Windows Autotools build
Fixes #912