Qucs/qucs #913: Can't configure on ubuntu 18.04 LTS

Phoveran, 2019-03-13:
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking for X... libraries , headers 
checking for gethostbyname... yes
checking for connect... yes
checking for remove... yes
checking for shmat... yes
checking for IceConnectionNumber in -lICE... no
configure: error: Qt >= 4.6.0 is required.
configure: error: ./configure failed for qucs

Keeps telling me this..... However I checked that I have Qt 5.9.5 installed. How to solve this?

guitorri, 2019-03-13:
Qt5 is not yet supported.

Phoveran, 2019-03-14:
> Qt5 is not yet supported.

Well...I tried Qt 4.8.7, not good as well~~

Rmano, 2019-03-14:
What is the problem exactly? I am using it on 18.04 lts as well, no problems here...

Phoveran, 2019-03-14:
> What is the problem exactly? I am using it on 18.04 lts as well, no problems here...

I downloaded qucs from Sourceforge, unzipped it and did ./configure , Then it tells me I have to install Qt >= 4.6.0. However I have Qt 4.8.7 installed.

Basilisvirus, 2019-03-19:
Is there any known fox for that? i cannot install it on my Ubuntu 18.04.2 LTS as well.

Rmano, 2019-03-19:
On 19/03/2019 10:01, Christianidis Vasileios wrote:
>
> Is there any known fox for that? i cannot install it on my Ubuntu 
> 18.04.2 LTS as well.
>
I just now pulled the "develop" branch on git, and on my system (18.04 
up-to-date) it configures ok. Compiling now.

Could you please report the exact error you have?

Romano

-- 
--
Romano at http://www.rgtti.com/

Phoveran, 2019-03-20:
> On 19/03/2019 10:01, Christianidis Vasileios wrote: Is there any known fox for that? i cannot install it on my Ubuntu 18.04.2 LTS as well.
> I just now pulled the "develop" branch on git, and on my [...]

Actually I can't figure out what your 'exact error' means... I just ran the configure file and it threw an error telling me that It needs Qt >= 4.6.0.

in3otd, 2019-03-20:
can you post the content of the file `qucs/config.log` (not the top-level `config.log`) and the output of `pkg-config --list-all | grep -i qt`

Phoveran, 2019-03-20:
> can you post the content of the file `qucs/config.log` (not the top-level `config.log`) and the output of `pkg-config --list-all | grep -i qt`

here is the file:
[config.log](https://github.com/Qucs/qucs/files/2986825/config.log)

and there is no output of that command

Rmano, 2019-03-20:
For a reference, I have: 

```[romano:~] % pkg-config --list-all | grep -i qt
Qt5DBus                        Qt5 DBus - Qt DBus module
QtXml                          Qtxml - Qtxml Library
QtDBus                         Qtdbus - Qt DBus module
QtScriptTools                  Qtscripttools - Qtscripttools Library
QtDeclarative                  Qtdeclarative - Qtdeclarative Library
Qt5Sql                         Qt5 Sql - Qt Sql module
QtGui                          Qtgui - Qtgui Library
QtScript                       Qtscript - Qtscript Library
QtDesigner                     Qtdesigner - Qtdesigner Library
QtDesignerComponents           Qtdesignercomponents - Qtdesignercomponents Library
Qt5PrintSupport                Qt5 PrintSupport - Qt PrintSupport module
Qt5Test                        Qt5 Test - Qt Unit Testing Library
Qt5Core                        Qt5 Core - Qt Core module
Qt3Support                     Qt3support - Qt3support Library
Qt5Svg                         Qt5 Svg - Qt Svg module
QtXmlPatterns                  Qtxmlpatterns - Qtxmlpatterns Library
QtTest                         Qttest - Qt Unit Testing Library
QtCLucene                      Qtclucene - Qtclucene Library
QtCore                         Qtcore - Qtcore Library
Qt5Xml                         Qt5 Xml - Qt Xml module
Qt5Network                     Qt5 Network - Qt Network module
QtSql                          Qtsql - Qtsql Library
Qt5Gui                         Qt5 Gui - Qt Gui module
QtHelp                         Qthelp - Qthelp Library
QtOpenGL                       Qtopengl - Qtopengl Library
Qt5OpenGL                      Qt5 OpenGL - Qt OpenGL module
QtUiTools                      Qtuitools - Qtuitools Library
Qt5OpenGLExtensions            Qt5 OpenGLExtensions - Qt OpenGLExtensions module
Qt5Widgets                     Qt5 Widgets - Qt Widgets module
Qt5Concurrent                  Qt5 Concurrent - Qt Concurrent module
QtSvg                          Qtsvg - Qtsvg Library
QtNetwork                      Qtnetwork - Qtnetwork Library
```

Rmano, 2019-03-20:
And: 
```
[romano:~] % wajig status libqt4-dev
Package                 Installed       Previous        Now             State
=======================-===============-===============-===============-=====
libqt4-dev          	4:4.8.7+dfsg-7ubuntu1	4:4.8.7+dfsg-7ubuntu1	4:4.8.7+dfsg-7ubuntu1	install
```
So I have installed QT 4.8.7

Phoveran, 2019-03-20:
> And:
>
> ```
> [romano:~] % wajig status libqt4-dev
> Package                 Installed       Previous        Now             State [...]

I checked mine... Qt 4.8.7 too.

 _Package                 Installed       Previous        Now             State
=======================-===============-===============-===============-=====
libqt4-dev          	4:4.8.7+dfsg-7ubuntu1	N/A            	4:4.8.7+dfsg-7ubuntu1	install_

in3otd, 2019-03-20:
but it's strange that `pkg-config` did not return anything about which Qt stuff is installed. `configure` uses `pkg-config` to know where the Qt stuff is located so if that does not work it will not be able to find any Qt.

Phoveran, 2019-03-20:
> but it's strange that `pkg-config` did not return anything about which Qt stuff is installed. `configure` uses `pkg-config` to know where the Qt stuff is located so if that does not work it will not be able to find any Qt.

So I used `sudo apt install qt4-default` to install qt4.8.7.  Is the problem coming from here?

Rmano, 2019-03-20:
I did use the commands as in the main README of the project: 
```
sudo apt-get install build-essential
sudo apt-get install libqt4-dev libqt4-qt3support
sudo apt-get install automake libtool libtool-bin gperf flex bison
```
...and it worked.

rjordans, 2019-04-13:
I just did a successful build on 18.04 here as well.  Is this still a problem?

closed 2019-04-15 by guitorri