Qucs/qucs #918: "Select" tool should be activated after dragging and dropping a component

Kezii, 2019-04-10:
I find annoying that after you drag and drop a component from the dock, you are left with no active tool, the left mouse click does nothing, and you have to click the icon in the toolbar to do anything, every time.
Why is the select tool not enabled by default?

in3otd, 2019-04-14:
um, the usual behaviour here is that after dragging and dropping a component from the dock it remains "attached" to the cursor and you can place it multiple times with a mouse left click; mouse right click rotates the component and middle click does nothing.
To exit this state you can simply press the *Esc* key and the *Select* tool will be enabled.
Do you see a different behavior on your system (Windows/Linux/etc.) ?

Kezii, 2019-04-14:
I'm not sure if I understand
There are two ways afaik to place a component:

1. Drag and drop, to place the component only once (this is what the issue is about)
2. Click the component, and click multiple times to place the component

Have you described method 2, or method 1 is broken on my system and it should behave like method 2?
I compiled qucs from sources on Linux

I made a video:
[2019-04-14 11-59-24.tar.gz](https://github.com/Qucs/qucs/files/3077249/2019-04-14.11-59-24.tar.gz)

in3otd, 2019-04-14:
yes, above I described the "method 2"; currently "method 1" does not behave much differently, you still need to press *Esc* to get back to the *Select* tool. I think this has "always" been the case (I mean, your system is not broken) but I see that it could be more useful if it went back to the *Select* tool or remembered the tool in use before the drag and drop.

ddrake, 2021-09-08:
Here's a [recent example from the forum](https://sourceforge.net/p/qucs/discussion/311049/thread/fe3cb418bc) that I think illustrates why the behavior should be to go back to the Select tool.