Qucs/qucs #920: Cannot Connect Wire to leds when they are in the vertical Position

cornellcummings, 2019-05-15:
I cannot connect a wire to any led when the object is rotated vertically. I noticed that when the component is horizontal, I can connect it. If I rotate it after the connection, it breaks the connection. I using version 0.0.19 on OS X mojave. The coordinate system also never really pinpoints the connection spot when vertical as well

in3otd, 2019-05-18:
Does this happen for every component?
Does selecting a component and pressing *Ctrl+U* (set on Grid) fix this?

cornellcummings, 2019-05-19:
I only used a few components but the main other one the resistor worked
fine. I'm not sure what Crtl+U does as I didn't notice anything different.
I tried selecting the component from the menu on the left ctrl+u and
selecting an existing component and ctrl+U.
This interesting thing is that it seems to be now working correctly. I can
rotate the LED to vertical and wire them Up. Rotating an object breaks the
connection though. But  rotating it back reconnects it. I don't think this
is a bug though and I think I might like it to do quick disconnections.
Since it's working I tried to make a new project and replicate what I did
as best I could and it is still working.

felix-salfelder, 2020-09-26:
closing as it seems to be now working correctly.

closed 2020-09-26 by felix-salfelder